MADEAPPS/newton-dynamics #16: Does not build using cmake on Linux

ColinGilbert, 2015-01-11:
Hi,

I am building Newton on Gentoo. However, I am stumped at:
```
/home/noob/CoolCode/newton-dynamics/applications/demosSandbox/sdkDemos/toolBox/toolbox_stdafx.h:106:19: fatal error: wx/wx.h: No such file or directory
 #include <wx/wx.h>
                   ^
compilation terminated.
applications/demosSandbox/CMakeFiles/demosSandbox.dir/build.make:57: recipe for target 'applications/demosSandbox/CMakeFiles/demosSandbox.dir/sdkDemos/DemoListenerBase.cpp.o' failed
make[2]: *** [applications/demosSandbox/CMakeFiles/demosSandbox.dir/sdkDemos/DemoListenerBase.cpp.o] Error 1
CMakeFiles/Makefile2:451: recipe for target 'applications/demosSandbox/CMakeFiles/demosSandbox.dir/all' failed
make[1]: *** [applications/demosSandbox/CMakeFiles/demosSandbox.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

```

It seems that wxWidgets is not being linked in correctly.

Manny-MADE, 2015-01-14:
Yes, it looks like the include directory is not properly setup. Unfortunately neither me nor Julio uses cmake (or Gentoo), so I can't look into it. Normally somebody starts a pull request with a fix like here: https://github.com/MADEAPPS/newton-dynamics/pull/14 But I don't think there should be an issue with wxWidgets as there hasn't been a change recently. Would be awesome if you could figure out why it's broken. 

But if everything else compiles you could still hack your generated makefiles and simply add the include directory.

dliw, 2015-01-20:
Are you sure you have wxWidgets header files installed? I don't know Gentoo, but usually these are -devel or -dev packages.

If you only need the library itself without the demosSandbox, you can change the define `NEWTON_DEMOS_SANDBOX`, e.g. via cmake-gui.

perkinslr, 2015-07-10:
Gentoo is a source based distro, so all the header files are always installed.  But it doesn't always put them where other distros do, and lets you select which versions of libraries to symlink in to the include path.  Check the output of `eselect wxwidgets list`, make sure you've got something selected.  I just built it on Gentoo using CMake.

closed 2017-05-08 by JulioJerez