MADEAPPS/newton-dynamics #26: No tag for v2.0

kyle-rader, 2015-08-28:
Hi,  I've been search around and I can't find a reference to where I can download v2.0 of the newton-dynamics library.  A tag in the source code to at least know where to checkout would be helpful.  If not something added to the wiki about how to get older versions required but some applications.  

Thank you!

JulioJerez, 2015-08-29:
Version 2.0 does no exist in Github. GitHub started at version 3.09
you need to get from Newton Dynamics Google source. but do it soon because
Google is shutting down Google source.

why do you want such an old version when you can get 3.13 here,

closed 2017-05-08 by JulioJerez

blitnicofo, 2020-02-14:
Hi @JulioJerez
We are porting an old game that works with Newton 2.08. 
Is there any way to access the source code of that version?

Thank you very much.

JulioJerez, 2020-02-14:
But I do not have anything lower than what is in GitHub.
When Google shutdown Google source I uploaded the current version to
GitHub, but I did not get the archive.
I suppose is possible to get it from old archive in Google but I do not
know how to do that.

The interface of the engine has change very little, updating to new version
is a matter of a day or two of work.