MADEAPPS/newton-dynamics #34: Unhandled exception in demosSandbox

Neill3d, 2016-05-09:
I've tried to drive a car in demosSandbox and got Floating-point invalid operation.
 Here is a screenshot with a call stack - https://drive.google.com/file/d/0B83XZ3TC_S6PeEYyXzVxeXhoZEE/view?usp=sharing

JulioJerez, 2016-05-09:
did you do something special? try sync and see if it can be reproduce it again.

JulioJerez, 2016-05-09:
I look at the line, it does no seem likely the code would crash there, since that a diagonal value of a matrix is always positive and larger than zero. Please sync and try again.

Neill3d, 2016-05-09:
I downloaded the latest master this morning, and build a sandbox from the scratch, I guess it should be sync. I'll try again.

JulioJerez, 2016-05-09:
oh cool, if you do that maybe you can make a replay, to detect when it happens.
basically you edit file Supercar.cpp anbd you change line  636 fro 
#if 0
	#if 0

To
#if 1
	#if 1

this will save the input to a file, to play replay you change the lines  to
#if 1
	#if 0

and will play the replay file, hopefully  it can reproduce the bug, and you can give to me.

Neill3d, 2016-05-09:
I've done a reply of a bug, I've uploaded log file on a google disk - https://drive.google.com/file/d/0B83XZ3TC_S6PRHFCNUtSSVNDaDQ/view?usp=sharing

P.S. I would like to say my feedback about car behaviour. I really like how steering works now, but when I try to drive forward or backward, looks like car is on the ice (low friction) - wheels rotate very fast, but movement is delayed and not so fast. This super car should run 0-100 much faster on the ground, I guess about 5-6 seconds.

JulioJerez, 2016-05-09:
thanks, the file recreates the bug.

and thank for the feedback, the car parameter are configurable . those are the parameter for the original 1992 Dodge Viper which is notoriously bad to drive.
but you are right I see if I can tide more.

JulioJerez, 2016-05-09:
Ha thank for that log file, it was very helpful on fix that bug.
I fix it already, it was a copy and paste mistake I made when exposing the internal solver to the engine tools. 
if you sync again it should be ok now.

Please if you test again, keep giving me feedback, the SDK becomes better when people provide  suggestion about what the want. otherwise is just my vision only.

Thank you.

Neill3d, 2016-05-09:
Yes, now a car demo works. Just found that a transmission is using only 2 gears to move forward.
I'll keep checking and let you know if I find something to say. Thank you so much for your project.

closed 2016-05-09 by Neill3d

JulioJerez, 2016-05-10:
It is using tow gears because I am still doing some calibration. I will add more control soon, and let you know. you are right there is too much power going from the  engine to the tire, and needs regulation.