MADEAPPS/newton-dynamics #35: master build failing on OS X

wivlaro, 2016-05-12:
There are two errors.

1. CLOCK_REALTIME being undefined
2. undeclared variable in dgSPDMatrix.h

Thanks.

```
[ 13%] Building CXX object coreLibrary_300/CMakeFiles/NewtonObj.dir/source/core/dgSPDMatrix.cpp.o
In file included from /Users/bill/Projects/newton-dynamics/coreLibrary_300/source/core/dgSPDMatrix.cpp:25:
/Users/bill/Projects/newton-dynamics/coreLibrary_300/source/core/dgSPDMatrix.h:117:19: error: use of undeclared identifier 'n'
        for (dgInt32 i = n - 1; i >= 0; i--) {
                         ^
/Users/bill/Projects/newton-dynamics/coreLibrary_300/source/core/dgSPDMatrix.h:119:31: error: use of undeclared identifier 'n'
                for (dgInt32 j = i + 1; j < n; j++) {
                                            ^
2 errors generated.
make[2]: *** [coreLibrary_300/CMakeFiles/NewtonObj.dir/source/core/dgSPDMatrix.cpp.o] Error 1
[ 14%] Building CXX object coreLibrary_300/CMakeFiles/NewtonObj.dir/source/core/dgThreadHive.cpp.o
[ 14%] Building CXX object coreLibrary_300/CMakeFiles/NewtonObj.dir/source/core/dgTree.cpp.o
[ 15%] Building CXX object coreLibrary_300/CMakeFiles/NewtonObj.dir/source/core/dgTypes.cpp.o
/Users/bill/Projects/newton-dynamics/coreLibrary_300/source/core/dgTypes.cpp:55:17: error: use of undeclared identifier 'CLOCK_REALTIME'
                clock_gettime(CLOCK_REALTIME, &ts);
                              ^
/Users/bill/Projects/newton-dynamics/coreLibrary_300/source/core/dgTypes.cpp:58:16: error: use of undeclared identifier 'CLOCK_REALTIME'
        clock_gettime(CLOCK_REALTIME, &ts); // Works on Linux

```

JulioJerez, 2016-05-12:
ok I fix the one at line 117 and 119, but I am not sure why  CLOCK_REALTIME is undefined, that's how people read teh system time in linux systems.

wivlaro, 2016-05-12:
I was building on Mac OS X

JulioJerez, 2016-05-12:
but this define CLOCK_REALTIME is compiled out for OS X, are you using xcode?
or is this linux in OS X?
I fixed other tow, it should build now

JulioJerez, 2016-05-12:
if this is linux for OS X then fidn out what is the right parameter to pass to clock_gettime in OS X

and I fix the bug, I can't o it because I never build Linux for Mac.

wivlaro, 2016-05-13:
The pull fixed the undefined 'n' yes, thanks.

I think I figured the other bit out. Looking at the coreLibrary_300/CMakeLists.txt it's using the "UNIX" check to set _POSIX_VER*:

```
if (${UNIX})
  if (BUILD_64)
    add_definitions(-D_POSIX_VER_64)
  else (BUILD_64)
    add_definitions(-D_POSIX_VER)
  endif (BUILD_64)
endif(${UNIX})
```

I changed it to this and it works: 
```
if(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
  add_definitions(-D_MACOSX_VER)
else(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
	if (${UNIX})
	  if (BUILD_64)
		add_definitions(-D_POSIX_VER_64)
	  else (BUILD_64)
		add_definitions(-D_POSIX_VER)
	  endif (BUILD_64)
	endif(${UNIX})
endif(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
```

According to https://cmake.org/Wiki/CMake_Useful_Variables#System_.26_Compiler_Information this is right.

wivlaro, 2016-05-13:
See PR #36

JulioJerez, 2016-05-13:
ha ok I added it to the cmake, 
please try again.
thanks

wivlaro, 2016-05-13:
Great! Thanks!

closed 2016-05-13 by wivlaro