MADEAPPS/newton-dynamics #39: Crash in superCar demo

Neill3d, 2016-05-26:
Hi,

 I've tested superCar from the latest sources (just uploaded today and compiled from scratch)
I couldn't reproduce the bug one more time. I was making a reverse 180 and got a crash - https://drive.google.com/file/d/0B83XZ3TC_S6PMnNCSkkyTzlmc0E/view?usp=sharing
 Also the demo works pretty slow, I have better performance when switch to "Iterative Solver Quality High" from the main menu.
 Anyway I like the handling, thank you, good improvment.

JulioJerez, 2016-05-27:
Oh think you testing.
The Assert on eh line does no necessarily mean the code will crash. all it is saying that that the 
volume of the  tetrahedrum is zero and the it should no calculate the inverse. 

What follow next is a test for zero, and the code proceed to the contingency as a special case. 
I left the assert for me to check if is in fact the fallback is resolve, but eh code will no crash if you continues.
you go the assert because you are running in debug mode. 

regarding the speed, it run slow because you are running in debug mode,  teh engine has lot of snanity check in debug mode, the lake very slow. 

Please compiled the release build, and it should run much better.
I get about 1000 or more fps, and my machine is no even that fast, is a 7 year old I core 7 Sandy bridge.

Again thank for trying the SDK.

Neill3d, 2016-05-27:
Thank you, now it's clear. And yes, release version runs well.
 But I guess I have something to show you. I've build a release_double version and from the first time tank hits the ground I've got a crash, I guess this time it's a bug - https://drive.google.com/file/d/0B83XZ3TC_S6PMFl0ck4wS2VSMEE/view?usp=sharing
 P.S. I was having several times situation when a "supercar" wheel partly goes inside the floor plane, but hard to catch, this is rarely happent.

closed 2017-05-08 by JulioJerez