MADEAPPS/newton-dynamics #41: Error Compiling dMath/dVector.h

AntonSynytsia, 2016-06-22:
I get an erro compiling dMath/dVector.h, on line 49:
`TemplateVector operator* (const TemplateVector &B) const {return (*this).CrossProduct(A);}`
Should it be:
`TemplateVector operator* (const TemplateVector &B) const {return (*this).CrossProduct(B);}`

JulioJerez, 2016-06-22:
Ha yes, I fixed already. please sync
I change the operator overload for dotproduct and cross product because I was having a hard time using wigg to make a wrapper for c#.
Swigg has ways to handle it, by is by renaming functions and it end up with  lot of compile errors.

also I want to do that long time ago, because the simple that I used do no realy obey the law of order of operations.

AntonSynytsia, 2016-06-22:
Ok thanks!

closed 2016-06-22 by AntonSynytsia