MADEAPPS/newton-dynamics #49: Default collision type for body must be NullCollision

KionX, 2016-11-03:
Default collision pointer is null.
The pointer must specify to a Null collision type
or calculations of the collision should be ignored.

JulioJerez, 2016-11-04:
is this a bug?

KionX, 2016-11-04:
this uncomfortable behaviour

JulioJerez, 2016-11-06:
Uncomfortable behavior, what does that mean I do not understand.

KionX, 2016-11-06:
If create body as
NewtonCreateDynamicBody(World, null, Matrix);
it throw exception in NewtonUpdate

the right to create so:
Collision=NewtonCreateNull;
Body=NewtonCreateDynamicBody(World, null, Matrix);
NewtonBodySetCollision(Body, Collision);

NewtonCreateNull meaningless, not needed

NewtonBodySetCollision(Body, null);
must be equivalent
Collision=NewtonCreateNull;
NewtonBodySetCollision(Body, Collision);

also CollisionNull has no parameters
it serialization is meaningless

JulioJerez, 2016-11-06:
oh I see you what to pass null for bodies passing null for null collisions.
I suppose that will be a short cut, by I prefect a uniform interface, each body must have a collision pointer 

NewtonCreateNull is not meaningless, the engine expect all bodies to have a non null collision shape, doing otherwise mean I will have to check all places when a collision is used an test for null, which can   can be several tenths of thousand of time during an update.

you must doing the second way
Collision=NewtonCreateNull;
NewtonBodySetCollision(Body, Collision); 

sorry for that inconvenient 
are you having problems with that and serialization?

KionX, 2016-11-06:
I had inconvenience because of this behavior
and I decided in a similar way:

NewtonCreateNull you can call NewtonCreate
and implicitly assign in
NewtonBodySetCollision and NewtonCreateBody

JulioJerez, 2016-11-06:
Yes I suppose I can do that on the engine side, but that makes the engine interface unnecessarily  complex for little benefit. Newton is suppose to be a low level library I do no like to add extra interfaces that is the end application can write a wrapper for it.
similar to that, people always ask for feature like stuff like forces at a point, and I do no add that to the engine because it can be done by using a cross product, 
this case is similar, you can write and wrapper function like 

MyCreateNewton (world, shape, matrix)
{
  if (shape == null)
 {
     shape = NewtonCreateNull();
     NewtonCreateBody(world, shape, matrix)
     NetwonDeletShape(shape)
} 
else 
{  
   NewtonCreateBody(world, shape, matrix)
}
} 

I actually do that for Collision shape matrices, but is a pain in the neck  to keep adding permutations of parameters. 
I hope this is Ok solution for you.

KionX, 2016-11-07:
It is unnecessary to add extra interfaces

I propose add to the class constructor dgWorld
m_nullShape = NewtonCreateNull();
and to dgBody::AttachCollision
if (!collisionSrc) {
    collisionSrc = m_world->m_nullShape;
}

JulioJerez, 2016-11-07:
ok, I just added that functionality,  please sync and let me know

KionX, 2016-11-07:
in general, this is what I wanted
but I don't understand why create a null collision for every body

I can propose my solution?

JulioJerez, 2016-11-07:
what do you mean it create a null collision for each body? it is not, 
it creates a null collision only when the pointer is null, The engine has internal cache system so in no making new collision.  you cna check that the pointer to the shapes is the same
this is the function

```
NewtonBody* NewtonCreateDynamicBody(const NewtonWorld* const newtonWorld, const NewtonCollision* const collisionPtr, const dFloat* const matrixPtr)
{
	TRACE_FUNCTION(__FUNCTION__);
	Newton* const world = (Newton *)newtonWorld;
	dgCollisionInstance* collision = (dgCollisionInstance*)collisionPtr;
	if (!collisionPtr) {
		collision = (dgCollisionInstance*) NewtonCreateNull(newtonWorld);
	}

	#ifdef SAVE_COLLISION
	SaveCollision (collisionPtr);
	#endif

	dgMatrix matrix (matrixPtr);
#ifdef _DEBUG
	//	matrix.m_front = matrix.m_front.Scale3 (dgRsqrt (matrix.m_front % matrix.m_front));
	//	matrix.m_right = matrix.m_front * matrix.m_up;
	//	matrix.m_right = matrix.m_right.Scale3 (dgRsqrt (matrix.m_right % matrix.m_right));
	//	matrix.m_up = matrix.m_right * matrix.m_front;
#endif

	matrix.m_front.m_w = dgFloat32 (0.0f);
	matrix.m_up.m_w    = dgFloat32 (0.0f);
	matrix.m_right.m_w = dgFloat32 (0.0f);
	matrix.m_posit.m_w = dgFloat32 (1.0f);

	NewtonBody* const body = (NewtonBody*)world->CreateDynamicBody (collision, matrix);
	if (!collisionPtr) {
		NewtonDestroyCollision((NewtonCollision*)collision);
	}
	return body;
}

```

KionX, 2016-11-07:
I used the wrong words.
Even internal cache system for them redundant.

JulioJerez, 2016-11-07:
not that will not work, when a body is created, the collision shape get its reference count incremented, 
assigning the pointer will cause a crash when the body is going to be destroyed.

KionX, 2016-11-07:
CreateNull - collision reference count +1
CreateDynamicBody (collision) - collision reference count +1
DestroyBody - collision reference count -1

I this so understand...

JulioJerez, 2016-11-07:
ok, now we agree

closed 2017-05-08 by JulioJerez