MADEAPPS/newton-dynamics #51: Separate collisions cache from world

KionX, 2016-11-09:
Need to use the same collision in several worlds.

Now I'm using the hack:
NewtonCreate
    С1 = CreateCollision
    C2 = CreateCollision
    ...

NewtonCreate
    NewtonCreateBody
        SetCollision( C1 )
    NewtonCreateBody
        SetCollision( C2 )
    ...

JulioJerez, 2016-11-11:
what do you mean using the hack? 
you do not really have to worry about keeping track of collision, the engine  keep track of that.
even  when you make  collision with different  scales. 

you can just do 
createcollsion 
createBody
Destroycollision.

each collision seems different, but that's only the instance, if tow collision are of the same dimension
say to box of size (A x B x C) then for you call 
void* NewtonCollisionGetSubCollisionHandle (const NewtonCollision* const collision);
you will see hat the handle is the same.

JulioJerez, 2016-11-11:
In Newton, worlds are like ship in the ocean that do not share anything in common. 
you can't not use anything from one world in another work.

if you need something like that, you need to add user data that allows you to have bodies in different world that are related by some common data stored in the user data.

KionX, 2016-11-11:
I need to use the same collision shapes in different worlds.

I'm creating NewtonWorld and creating collisions.

Next, create another NewtonWorld.
In it, create the bodies. Assign them collisions from first NewtonWorld.
Perform simulate physics.

It works with no problem.
But think to use the hack is not a good idea.

JulioJerez, 2016-11-11:
that will create lot of memory leaks. because the manager put cookie information on assets.
The engine does not support moving asset form one world to the next.
you nee to make a copy.

a quick way to do it is to serialize the object from one world to memory, and the reserialized it onto the other world. that will crate the proper manager.

JulioJerez, 2016-11-11:
what are you doing? maybe I can suggest some ideas.

KionX, 2016-11-11:
I'm making a server where there is a lot of independent physical simulations using the same assets.

I don't moving collisions between worlds. They remain where it was created.
Reference counting is not disturbed...
How can it leak memory?

closed 2017-05-08 by JulioJerez