MADEAPPS/newton-dynamics #52: Crash due to stack exhaustion in worker threads when creating a large number of bodies

isegal, 2016-11-27:
When creating a large number of moving objects (i.e. >4000) default thread size is too small and the system will crash.

When having no worker threads, this can be fixed by passing a larger thread size when creating newton world with NewtonCreateEx().

But if we also set NewtonSetThreadsCount(), the system will crash inside worker thread due to stack exhaustion.

Increasing stack size for worker threads can only be done by editing DG_WORKER_TRAD_STACK_SIZE_IN_BYTES in the source and it solves the problem however it's not ideal because newton source code has to be modified.

Proposed solutions:

Option 1:
Allow worker threads to use the same thread size setting as passed for main thread during world creation.

Option 2:
Add an additional parameter for worker thread stack size in either NewtonCreateEx(), or NewtonSetThreadsCount() or a new function such as NewtonSetThreadStackSize()

Thank you.

godlikepanos, 2016-11-28:
If I'm not mistaken this problem is caused by alloca(). IMHO alloca is a very dangerous function and it should be avoided. 

What makes sense to have is a linear allocator. This linear allocator has an initial size and it grows if it needs to. When allocating you just increase an atomic and you only need to lock if you need to grow it (that makes it thread-safe). At the beginning of the frame you just move the atomic to the beginning of the pool to reset it so you can re-use it. After a few frames it has grown and there will be no locks.

The annoying thing with the linear allocator is that you have to carry it around.

Julio, if you want an implementation reference check this the StackMemoryPool class in this https://github.com/godlikepanos/anki-3d-engine/blob/master/src/anki/util/Memory.h and https://github.com/godlikepanos/anki-3d-engine/blob/master/src/anki/util/Memory.cpp

JulioJerez, 2016-11-29:
this should not happens anymore in 3.14, I fixed last week.

JulioJerez, 2016-11-29:
NewtonCreateEx should not be necessary anymore. Please since to latest version and let me know if there is still a problem wit that.
are you have scene with that many objects?  is this a game? 
also about the alloca, yes I know this could be dangerous but of the other hand is very handy for when a function required use variable size stack variable. 
Use allocation end up is terrible memory fragmentations,  and use linear allocation can sole the problem by you need to keep track of threads and so that you know which pool to do the allocation form. 
Is possible to use global thread memory, but the you run into the problem of global variables that are local to some tread and so on and so forth.
I prefer to state away from the level of complexity unless is absolutely needed. 

again, please  sync to later 3.14 and let me know of this is still a problem.

closed 2017-05-08 by JulioJerez