MADEAPPS/newton-dynamics #60: Bulding from the latest sources

Neill3d, 2017-02-21:
From some time I have a problem in building sandbox demo from sources. I'm using VS 2010
 Problem starts from the projects paths in a solution, for example
D:\temp3\newton-dynamics-master\packages\projects\visualStudio_2010\dMath.vcxproj
 which is wrong and I have to manually switch to
D:\temp3\newton-dynamics-master\packages\projects\visualStudio_2010_dll\dMath.vcxproj

Then many errors during compilation appear, like that
5>..\..\sdkDemos___\toolBox\PhysicsUtils.cpp(915): error C3861: 'NewtonCollisionGetSkinThickness': identifier not found
5>..\..\sdkDemos___\toolBox\PhysicsUtils.cpp(1235): error C2660: 'NewtonBodyAddImpulse' : function does not take 3 arguments

and etc.

What am I doing wrong ? I remember times (around half a year ago) when everything was fine during the build process.

JulioJerez, 2017-02-22:
do you have the later version.

Neill3d, 2017-02-22:
I've downloaded from main repository and master branch - https://github.com/MADEAPPS/newton-dynamics
 I was doing like that before and it was fine to build from sources... but now it doesn't work, too many errors. That's why I'm a bit confused how to use and check your updates.

JulioJerez, 2017-02-22:
what errors are you talking about? 
the build should compile with zero warning and zero errors on all version of visual studio.

Neill3d, 2017-02-27:
Hi, I'm using VS 2010. I also have VS 2015 and have the same issues there.
 
[Here is a screenshot of what I have when starting up the demosSandbox.sln](https://drive.google.com/file/d/0B83XZ3TC_S6PNGgzUzJoaW5ldnM/view?usp=sharing)
 
[And solution output log](https://drive.google.com/file/d/0B83XZ3TC_S6PNmVMaVAwY2xKR1U/view?usp=sharing)
[Then I'm doing path correction like according to a project folder names](https://drive.google.com/file/d/0B83XZ3TC_S6PbTdBZmt3UFVOaFU/view?usp=sharing)

 Even after correction path step is over and I have my solution projects loaded, I have batch of errors while building a solution... like I've shown in my first message. But I didn't try to build in the VS 2015.
 Is the VS 2010 is not supported anymore ?

JulioJerez, 2017-02-27:
you are opening the wrong project, for vs 2010 the right project is 
C:\Users\jjerez\Downloads\NewtonSDK\newton-dynamics\applications\demosSandbox\projects\visualStudio_2010\demosSandbox.sln

Neill3d, 2017-02-28:
Now it works, thank you. Sorry for taking your time for that.
I was thinking that imgui is what you were integrating not long time ago and it should be the right project.
 Are you still using imgui in sandbox or it's not fit your needs for some reason ?

JulioJerez, 2017-02-28:
imgui is great and aweseme, and I will move to that eventually, but for now we still use wxwidget until we get pass some o fhe issue with the engine an try again to build the mac.
WxWidget is also very good, but is too big for making simples demos and requires lot of set up each time they make a new version.

Neill3d, 2017-03-01:
I see, thank you

closed 2017-03-01 by Neill3d