MADEAPPS/newton-dynamics #64: VS2015 builds Preprocessor Definition

AntonSynytsia, 2017-03-20:
When building newton on VS2015 with all static and dynamic libraries set to
_Visual Studio 2015 - Windows XP (v140_xp)_ Platform Toolset rather than just _Visual Studio 2015 (v140)_, the compiler outs an error saying something like:
C++ warning C4005: '__useHeader': macro redefinition

To fix it the answers here, http://stackoverflow.com/questions/14363929/vs2012-c-warning-c4005-useheader-macro-redefinition, suggest appending `%(PreprocessorDefinitions)` to the Preprocessor Definition property. I appended that marco to all the projects and it  fixed the building error. So, I think that to make all VS2013 and VS2015 projects compatible with _Visual Studio 2015 - Windows XP (v140_xp)_ platform toolset, all of them must have `%(PreprocessorDefinitions)` appended to Preprocessor Definitions property.

JulioJerez, 2017-03-20:
but that is a warning, isn't? the project should still compile since the engine does no use anything that is OS depended.
That's a lot of work over too many projects configurations, I  added this line to the include file
#pragma warning (disable: 4005) //'__useHeader': macro redefinition

that should prevent the warning

AntonSynytsia, 2017-03-20:
Yes that's a warning, but for physics.vcproj all warnings are set to be treated as errors.

Thanks for adding that thing that prevents the warning.

closed 2017-03-20 by AntonSynytsia