MADEAPPS/newton-dynamics #65: dgAssert(0)

AntonSynytsia, 2017-05-04:
There is a dgAssert(0) in the dgWorldDynamicSimpleSolver.cpp, line 318. Is it necessary there?

JulioJerez, 2017-05-04:
no it is no necessary, I will remove with next commit. 
are you doing CCD?

AntonSynytsia, 2017-05-05:
Yes, exactly.

The assert occurred when CCD for a body was enabled, but in my AABB overlap callback I returned 0 because I intended the body to be non-collidable.

I think that in my implementation it would be right to automatically disable CCD for a body whenever it's set to non-collidable. However, I don't know if this fixes the issue of other bodies, with CCD enabled, colliding into non-collidable bodies. The AABB callback will be called, but because I return 0, and CCD is enabled for the collidable one, there should be an assert.

Perhaps the CCD should consider what the custom AABB overlap callback returns before proceeding to computing.

AntonSynytsia, 2017-05-05:
If the ABBB callback returns zero, then the CCD should not be computed for two AABB overlapping bodies.

JulioJerez, 2017-05-05:
it is fine to let that CCD, thai way I get more testing.  CCD has automation checks so it is only spensive wh it really need to the work.

on this 
> The assert occurred when CCD for a body was enabled, but in my AABB overlap callback I returned 0 because I intended the body to be non-collidable.<
is not necessarily true, imagine two one if from of another, and another hitting the from one at high speed. if the aabb return false on the tow object that are not moving fast relative to each other, after the first hit, on of the two bodies will gain lot of speed, but since time step is not complete it is possible that the second body go through the on in front because it did no see it at all.

but yes is the AABB test return zero, then that collision pair is not part of the island.

AntonSynytsia, 2017-05-05:
Okay, that's good! Waiting until the are other asserts with CCD are fixed

AntonSynytsia, 2017-05-05:
There is also another assert, in dgWorldDynamicUpdate.cpp, line 831, which I think shouldn't be there:
dgAssert (!body1->m_equilibrium);

Here is when it causes issues:
Say a model has bodyA that is connected to jointA and bodyB that is connected to jointB. The assert is called, if bodyB is sleeping and bodyA comes into contact with bodyB.

Commenting out the assert doesn't seem to cause any issues.

So, should it be removed?

JulioJerez, 2017-05-05:
oh that's a more serous one that should not happens, when a body come into contact with another that is sleeping then the entire set should be considered alive because sleep flag of the cluster is the and operation of the union of all bodies on the cluster.

commenting it out may hide a bug, can you serialize the scene that reproduce and send it to me for debugging?

closed 2017-09-23 by JulioJerez