MADEAPPS/newton-dynamics #68: Forum registration issue / questions regarding suitability of newton dynamics

eabbink, 2017-06-21:
Hi,

as I could not find an alternative way to communicate I am trying here, even though it might not be the most appropriate location. The site/forum says to contact admins when having trouble with registration but as far as I could find it does not provide a way to actually reach said admins..

As I have a couple of questions about newton before I start integrating it into our environment I tried to register on the forum. I have tried to register using two different addresses, and I have clicked the "resend activation email" link for both, but in neither case I am seeing any emails turn up. (spam folder double checked as well). This concerns user registrations 'eabbink' and '421safari'. If an admin could activate (one of) them that would be great.

Now for the questions I have regarding Newton Dynamics. We have our own vehicle model for which several new requirements (mainly joints and increased stability) make it preferable to incorporate an existing physics engine. Accuracy and stability valued above performance. To be able to (easily) integrate Newton (or any other engine) we need the following:

1. double precision support (related to large worlds): _This should be present in Newton, however according to some old posts it may not complete/functional or even buildable. Enabling the option in cmake builds fine however, and the demos application runs fine as well. so OK?_

2. step the simulation a certain amount of time: _Call NewtonUpdate ()_

3. for a rigid body query position: _Call NewtonBodyGetPosition or NewtonBodyGetMatrix_

4. for a rigid body query orientation: _Call NewtonBodyGetRotation or NewtonBodyGetMatrix_

5. for a rigid body query linear velocity: _Call NewtonBodyGetVelocity_

6. for a rigid body query linear acceleration: _No function present?_

7. for a rigid body query angular velocity: _Call NewtonBodyGetOmega_

8. for a rigid body query anguler acceleration: _No function present?_

9. multiple times between simulations steps, add a force at a position (force may generate resulting torque): _Callback mechanism is inconvenient here, but could be worked around by storing a list of forces to sum in the callback. For NewtonBodyAddForce() the manual lists: "**This function is only effective when called from NewtonApplyForceAndTorque callback**". Why is this the case? Why couldn't one simply call this function a number of times adding forces before calling NewtonUpdate?_ 

10. dynamic joint deletion and creation between existing rigidbodies.

11. live changing of parameters (mass, stiffness, etc).

So at first glance most of the basics seem ok except for 6 and 8. Is it really not possible to query accellerations for a rigidbody? Note that for our application simply deriving them from velocities is a nogo.

Thanks!

Esger

JulioJerez, 2017-06-21:
I activated you account eabbink by hand, I do no know what is wrong with
the forum, few people has had that problem and I do not know how to fixed
it.
anyway if you post that comment over the forum, over there is a better way
to have a conversation.

eabbink, 2017-06-21:
Thanks!

agreed, I've posted the questions on the forum.

closed 2017-09-23 by JulioJerez