MADEAPPS/newton-dynamics #69: updateAutomaticGearBox issue

velkyel, 2017-06-26:
At line https://github.com/MADEAPPS/newton-dynamics/blob/master/packages/dCustomJoints/dCustomVehicleControllerManager.cpp#L1109 is m_gearsCount limited to 4.  Why?

btw: I tried to register at forum (login: velkyel) but I didn't received confirmation email, can you please check it out?

JulioJerez, 2017-06-26:
> packages/dCustomJoints/dCustomVehicleControllerManager.cpp#L1109 is

no sure why the forum does no send confirmation, I activated your account
manually, you can post now.

JulioJerez, 2017-06-27:
no the is no reason why I was testing the vehicle drift functionality, for that according to vehicle dynamics equations you need a steady velocity, and with automatic transmission the only way to go that is to limit the gear to firs and secund.

otherwise the vehicle will try to change gear  making drift much difficult.

you can comment the line out and it will go up all the way up to the fifth gear.

 

Thank for checking it out

Julio

closed 2017-06-30 by velkyel