MADEAPPS/newton-dynamics #70: Not Receiving Forum Activation Code

sdarkwell, 2017-07-14:
Hello!

I've registered for the forum (using two different Email addresses at different domains), and have yet to receive the required code.

I was hoping to post regarding difficulty compiling the NewtonUnityPlugin--I've been quite keep at looking into it, both due to personal interest in using this fantastic physics project, but also to see if any of my Editor/Inspector coding experience might be able to assist with the porting.

If you are willing to approve my request, that would be fantastic. If you decide to do so, please approve my Email @darkwellstudios, as that is my code-related one.

Thank you so much in advance!
- S.

JulioJerez, 2017-07-14:
Hi, 

 

I do not know what is wrong with the forum. 

 

what is your user name?  so that I can approve it.

 

Julio

sdarkwell, 2017-07-14:
Thank you for the quick reply!

I have requests under two different usernames:
"S. Darkwell"
-and-
"S.Darkwell" (no space)

If you can approve the username with the space, that is the one connected to the Email I prefer. Please feel free to remove the other.

Thank you again!
- S.

JulioJerez, 2017-07-14:
ok I activate "S. Darkwell"

 

at some point the second will be remove since account with not post are deleted periodically.

sdarkwell, 2017-07-14:
Many thanks, sir!

I will post on your forum ~6 hours from now when I return. It would be fantastic to resolve these compilation issues so I can experiment with your plugin over the weekend!

Thank you again. :)
- S.

JulioJerez, 2017-07-14:
ok

and thank for registering.

Maybe you will be the one trying the new techno I am developing self balancing characters the I believe will be ahead of anything out there 

Unity seem a good platform to show case it.

 

looking forward to see you in the forums.

Julio

closed 2017-09-25 by JulioJerez