MADEAPPS/newton-dynamics #74: CMake configure on fails

hhyyrylainen, 2017-08-05:
After updating to the latest commit cmake fails to run with this error:
```
CMake Error at CMakeLists.txt:108 (add_subdirectory):
  add_subdirectory given source
  "/home/hhyyrylainen/Projects/leviathan/ThirdParty/newton-dynamics/packages"
  which is not an existing directory.

CMake Error at CMakeLists.txt:109 (add_subdirectory):
  add_subdirectory given source
  "/home/hhyyrylainen/Projects/leviathan/ThirdParty/newton-dynamics/packages/thirdParty"
  which is not an existing directory.

CMake Error at CMakeLists.txt:110 (add_subdirectory):
  add_subdirectory given source
  "/home/hhyyrylainen/Projects/leviathan/ThirdParty/newton-dynamics/coreLibrary_300"
  which is not an existing directory.

-- Configuring incomplete, errors occurred!
See also "/home/hhyyrylainen/Projects/leviathan/ThirdParty/newton-dynamics/build/CMakeFiles/CMakeOutput.log".
```

Which looks like to have been caused by those missing directories to have been deleted at some point.
Was this intentional, and the cmake script didn't get updated, or should those directories still exist.

I'm now using commit `7c5970ccda537dea134e0443d702ef9f5ce81a38` which has those directories and compiling works with that version.

I also tried running git bisect but that didn't work well as there are bunch of commits in master that don't compile.

JulioJerez, 2017-08-05:
yes I am making a directory structure reorganization.  
give me a few it is still under changes, please give few days to finish it.

dliw, 2017-08-29:
This issue should be solved now.

closed 2017-09-25 by JulioJerez