MADEAPPS/newton-dynamics #79: ConvexCast broken with triangle meshes / Plz activate forum account.

rjappy, 2017-09-01:
hi,
 NewtonWorldConvexCast is not returning any contacts when tracing against certain triangle meshes.
i included a simple test cast.

--------------------
 
#include <stdio.h>
#include "Newton.h"
#include "dVector.h"
#include "dMatrix.h"

using namespace std;

void test() {
	NewtonWorld *world = NewtonCreate();

	// create static body
	NewtonCollision* cs = 0;
	NewtonBody* dyn = 0;
	{
		// Dummy transform matrix.
		dMatrix tm(dGetIdentityMatrix());
		#if 0
		// WORKING simple BOX
		cs = NewtonCreateBox(world, 40., 40., 0.2, 0,0);
		#else
		// TRIANGLE MESH
		dVector verts[6] = {
			#if 1
			// FAILING,TRIANGLE MESH
			dVector(-55.12000656,-38.16000366,0.00000000),
			dVector(55.12000656,-38.16000366,0.00000000),
			dVector(-55.12000656,38.16000366,0.00000000),
			dVector(55.12000656,-38.16000366,0.00000000),
			dVector(55.12000656,38.16000366,0.00000000),
			dVector(-55.12000656,38.16000366,0.00000000)
			#else
			// WORKING,TRIANGLE MESH
			dVector(+40,+30,0),
			dVector(-40,+30,0),
			dVector(-40,-30,0),
			dVector(-40,-30,0),
			dVector(+40,-30,0),
			dVector(+40,+30,0),
			#endif
		};
		
		cs = NewtonCreateTreeCollision(world,0);
		NewtonTreeCollisionBeginBuild(cs);
		NewtonTreeCollisionAddFace(cs,3,&verts[0][0],sizeof(dVector),0);
		NewtonTreeCollisionAddFace(cs,3,&verts[3][0],sizeof(dVector),0);
		NewtonTreeCollisionEndBuild(cs,0);
		#endif
		dyn = NewtonCreateDynamicBody(world, cs, &tm[0][0]);
		NewtonDestroyCollision(cs);
	}
	
	// cast shape
	NewtonCollision* cast_shape = 0;
	{
		cast_shape = NewtonCreateBox(world,4,4,16,0,0);
	}
	
	// convex cast
	dVector start(-10,20.5,9.5);
	dVector target(-10,20.5,5);
	dMatrix start_xform(dQuaternion(),start);
	float hitParam = 0;
	NewtonWorldConvexCastReturnInfo info[16];
	int contactCount = NewtonWorldConvexCast(world,&start_xform[0][0],&target.m_x,cast_shape,
											 &hitParam,0,0,
											 info,sizeof(info)/sizeof(info[0]),0);
// 	assert(contactCount > 0);
// 	int a = sizeof(dVector);
// 	dVector v;
// 	int b = sizeof(v.m_x);
// 	printf("%d %d ",a,b);
	
	
	printf("%f %d\n",hitParam,contactCount);
	
	// Tear down.
	NewtonMaterialDestroyAllGroupID(world);
	NewtonDestroyCollision(cast_shape);
	NewtonDestroyCollision(cs);
	NewtonDestroyAllBodies(world);
	NewtonDestroy(world);
}

--------------------

also please activate my forum account , account name is 'marco'
looks like if the email id has any numbers it gets flagged as spam.

thanks

JulioJerez, 2017-09-23:
ha than for that bug report, I pasted the demo in the sandbox and yes I can see that they is a bug there.
I am fix it now.
you account is activated.

Thank for the bug report

JulioJerez, 2017-09-23:
ok I see the bug, I will fix this afternoon, it is a rounding error bug, what happen is that the second test is the one that is failing because the box hit the cats is hitting right at the surface. 
the new newton this contacts are considered rejected.
the second because the face are inter floats, the contact are exact value, by that should rejected by the contact calculator. So that is in fact a bug.
The solution is that the convex cast should pad the bug with a tolerance so that if find a intersection point, then the contact will return a hit.
what I am curious is who a exact contact is report as a hit, that should not happens because the collision system in newton 3.14 use exact adaptive integer precision arithmetic, so collision should be right 100% of the time. 
anyway is a very simple fix.

if you are reading this, can you please respond in the forum
you account is active now.

JulioJerez, 2017-09-24:
Ok I have this fixed now.
Please sync and let me know if is work for you.
I committed the sandBox set to demo use, 
...\newton-dynamics\applications\demosSandbox\sdkDemos\demos\UsingNewtonMeshTool.cpp
where I pasted you repro test, if you find anything you can edit that demo to reproduces.

This was a serious bug,  no sure how it went undetected for so long.
anyway thanks for the repro test.

JulioJerez, 2017-09-25:
Oh, there is still a bug, It will pass your test, by is fail on other case. 
I set the sandbox demo to the convex cast demo and I can see where the call fail on oblique cast.
I will fix to tonight.

JulioJerez, 2017-09-27:
ok this was fixed with later commit.

closed 2017-09-27 by JulioJerez