MADEAPPS/newton-dynamics #80: Something wrong with here >> not building

FreeFromItAll, 2017-09-22:
https://github.com/MADEAPPS/newton-dynamics/blob/4c77872373fba6ef8e460614fa7cbe91f351ddb2/sdk/dgCore/dgGeneralMatrix.h#L63

JulioJerez, 2017-09-23:
is this a bug? 
can you tell me what is the compiler error please?

FreeFromItAll, 2017-09-25:
Actually, look at the line referenced, I presume you meant a function call
there rather than a declaration... Probably just a copy/paste error.

JulioJerez, 2017-09-25:
oh I see, not that was not a function call, I usually copy that prototype of a function when I am writing the inline version. I forget to either commented out or delete the line.
fixed, Thanks for that 
Julio

FreeFromItAll, 2017-09-25:
Glad I could help! Thanks for this awesome library. I'll update my local
version then.

closed 2017-09-25 by JulioJerez