MADEAPPS/newton-dynamics #82: Compiling issues.

volcoma, 2017-12-02:
Hi. I have several compiling issues. I am trying to build newton in my project with cmake.
I am using Qtcreator as an IDE on Windows. CMake is version 3.9
-mingw - cannot compile at all. Getting compiler errors in dgVector.h  about not finding __m128 type. Maybe some defines weren't properly set when using mingw.
-msvc - inconsistent dll linkage on mostly everything. I am Maybe the _NEWTON_STATIC_LIB isn't set properly.

TrevorCash, 2017-12-03:
I get this problem as well

volcoma, 2017-12-06:
Should we expect any fixes on this issue since I saw It in the forums as well which was a while ago. I would like to use newton as a submodule so I wouldn't like to change the internal cmakelists

dliw, 2017-12-15:
For Windows you have to use the Visual Studio project files for now.
The CMake project contains some Windows specific settings, but they are taken from an much older structure and have never been tested on Windows since. Patches that improve on this are very welcome.

Btw. currently there is also problem with the CMake project on Linux. I haven't had the opportunity to create a patch yet.

volcoma, 2017-12-20:
That's too bad. So basically the library is not in a useable state since the integration part is not operational. I wanted to use newton in my game engine since I've used it before and I am a fan, but I guess I will either have to wait or go with another choice.

TrevorCash, 2017-12-20:
A good candidate for integration for example would be https://urho3d.github.io/ The engine already implements alot of third party open source projects with cmake.

JernejL, 2018-01-15:
@volcoma - you can link to the dll interface, sadly this won't give you object oriented api, but still fully functional newton. 
 
I will try to get it to build with mingw and try to fix makefiles eventually. 
 
Also, msvc builds work fine - what is giving you problem? it compiles, at least the 2013 version is fine.

volcoma, 2018-01-15:
well just tried it again and its even worse. Still getting a huge amount of  warnings like
```
D:\Workspace\newton-dynamics-master\sdk\dgNewton\Newton.cpp:102: warning: C4273: 'NewtonGetMemoryUsed': inconsistent dll linkage
```
-It's basicly on every function which has NEWTON_API. 
Also this 
```
D:\Workspace\newton-dynamics-master\sdk\dScene\dBoneNodeInfo.cpp:22: error: C1083: Cannot open include file: 'tinyxml.h': No such file or directory
```
I am building only the sdk with turned off examples with the current master version

JernejL, 2018-01-15:
_NEWTON_STATIC_LIB should NOT be defined
_NEWTON_BUILD_DLL should NOT be defined
_WIN32 should be defined aswell

are these defines properly set in your project? this should lead to defining NEWTON_API as "#define NEWTON_API __declspec (dllimport)"
 
That would make the header import those properly.. from my understanding:

#ifdef _NEWTON_STATIC_LIB
	#define NEWTON_API
#else 
	#ifdef _NEWTON_BUILD_DLL
		#ifdef _WIN32
			#define NEWTON_API __declspec (dllexport)
		#else
			#define NEWTON_API __attribute__ ((visibility("default")))
		#endif
	#else
		#ifdef _WIN32
			#define NEWTON_API __declspec (dllimport)
		#else
			#define NEWTON_API
		#endif
	#endif
#endif

Also, you'll need correct _MINGW_32_VER or _MINGW_64_VER set aswell - check that those are ok

JernejL, 2018-01-15:
Maybe this helps aswell: http://newtondynamics.com/forum/viewtopic.php?f=12&t=4874#p34818 
 
 ricardonho and  agi_shi were succesfully using mings / code::blocks to link to dll succesfully.

volcoma, 2018-01-15:
Well I did not set anything. I downloaded the library as it is from GitHub and opened it with qtcreator as it supports cmake

dliw, 2018-01-15:
>as it supports cmake

Yes, but only tested on Linux.

For Windows currently only the Visual Studio project files work:
https://github.com/MADEAPPS/newton-dynamics/tree/master/sdk/projects

Or you could try to improve the CMake project ...

TrevorCash, 2018-01-15:
Maybe a fork should be setup to facilitate Modern cross-platform cmake integration.

tly000, 2018-01-19:
i compiled it yesterday on mingw64 (msys2) and it worked "okayish" :) you just need to manually fix some compile errors. to fix the `__m128` errors, you need to `#include <x86intrin.h>` in `dgVector.h`.

nevertheless the library doesnt work yet, i get a sigsegv on `NewtonCreate` ...

JernejL, 2018-01-22:
tly000: can you share the makefile you used? i'd love to try this with 32 bit, maybe i can help out and help make progress.

tly000, 2018-01-23:
i used the cmake file to build it. but i 7 other files to make it compile and not crash on startup. can i post some kind of diff here?

JernejL, 2018-01-24:
Sure, you can make a diff and attach it to this issue, so does your makefile fully work - no longer crashes during newtoncreate?

tly000, 2018-01-24:
I made a fork and pushed the changes that i needed to make it compile. I still have the crash on `NewtonCreate`, but i know a fix for that: Removing the custom allocator code and replacing it by simple `malloc` and `free` calls... this is of course not a good solution.

https://github.com/tly000/newton-dynamics/commit/5c7188a3501dc7d3b0fcf80a5b22a63dc70ecebb

closed 2019-03-31 by JulioJerez