MADEAPPS/newton-dynamics #83: Next release?

DrLynix, 2017-12-04:


JulioJerez, 2017-12-04:
I am getting ready for a final release by the end of the month.

DrLynix, 2017-12-04:
Glad to hear that!

Also, I tried to make myself an account on Newton forum, but I can't activate it since I never received the activation mail (not even in my spam folder). Is it possible that mail sending is currently broken?

JulioJerez, 2017-12-04:
Ha yes the forum app is broken for a long time and people can't register. It needs to be updated.
if you tell me you user name I can active the account.

DrLynix, 2017-12-05:
Sure, my username is "Lynix".

Should I use the forum to ask some questions about how to use Newton, or maybe you prefer GitHub?

JulioJerez, 2017-12-06:
you account is active now.
you can use the forum, that way more people provide different answers and also can benefit of the discussion.

DrLynix, 2017-12-06:
Alright, thank you!

closed 2017-12-06 by DrLynix