MADEAPPS/newton-dynamics #85: Registration to the forum does not work

Tymolc, 2017-12-19:
Hi, I am using a ruby wrapper of newton dynamics in Sketchup (it's called MSPhysics). I wanted to ask some questions regarding your engine in the forum, but I don't seem to be able to register (I don't get an confirmation link).
Is there something you can do?

JulioJerez, 2017-12-19:
go Daddy hosting is down, I notified them and they said will be restored with 72 ours.
They said I can fix it myself but I know nothing about web hosting so I have not choise but to wait.

Meanwhile you can post your questions about the engine here. 
Notice I am not the author of the MSPhysics plugin.

JulioJerez, 2017-12-20:
ok the site is back on.
can you tell me you user name so that I can active your account.
There is still a problem wi th registration that has not been solved yet.

JernejL, 2018-01-15:
@Tymolc - Email should work again now, but you will need to tell which account you registered so that we manually activate it.

JernejL, 2018-01-15:
@Tymolc - found your username and activated it.

Tymolc, 2018-01-15:
Oh, right. I'm sorry. Totally missed the replies. Thanks, it is working.

closed 2018-01-15 by Tymolc