MADEAPPS/newton-dynamics #89: NewtonConstraintCreateHinge implementation missing

execomrt, 2018-02-17:
The implementation is missing in Newton.cpp, but it is declared in the Newton.h.

execomrt, 2018-05-17:
NewtonJoint* NewtonConstraintCreateHinge(const NewtonWorld* const newtonWorld, const dFloat* pivotPoint, const dFloat* pinDir, const NewtonBody* const childBody, const NewtonBody* const parentBody)
{
	TRACE_FUNCTION(__FUNCTION__);	
	Newton* const world = (Newton *)newtonWorld;
	dgBody* const body0 = (dgBody *)childBody;
	dgBody* const body1 = (dgBody *)parentBody;
	dgVector pivot(pivotPoint[0], pivotPoint[1], pivotPoint[2], dgFloat32(0.0f));
	dgVector pin(pinDir[0], pinDir[1], pinDir[2], dgFloat32(0.0f));
	return (NewtonJoint*) world->CreateHingeConstraint(pivot, pin, body0, body1);
}

TrevorCash, 2018-08-23:
I Believe this portion is depreciated and may be removed in favor of using the custom joints api.

JulioJerez, 2019-01-07:
is is deprecated because the joint do lot of stuff, the can be more, have friction, be spring, have limits, be a servo, and so on.
all that functionality crate a combination set of functions or parameters. that are hard to support.
by moving the to the custom joint they do what they need to do disregarding other functionality.

what are you doing that you need the specific joint? 
are you on the latest version?

marchevskys, 2019-01-08:
@JulioJerez ok, I understand you. I am discovering joints with Newton.
Want to do advanced vehicle steering (Ackerman mechanism) with caster angle, etc.
Now I found UserJoint with arbitrary amount of DOF-s. Experimenting, how to work with it.

closed 2019-03-31 by JulioJerez