MADEAPPS/newton-dynamics #91: _mm_cmpeq_epi64 used in TestZero even when __SSE4_1__ isn't defined

leetNightshade, 2018-03-09:
__SSE4_1__ isn't commonly defined by compilers, and your function TestZero relies on _mm_cmpeq_epi64 and thus fails to compile on Linux using latest of clang. 

Also it doesn't check against proper macros for Unix/Linux systems, _POSIX_VER is defined in unistd, which is included after your check. I noticed on Mac Darwin you define _POSIX_VER in the cmake file, instead of properly checking for it in code.

JulioJerez, 2018-03-09:
oh you are right I did not know _mm_cmpeq_epi32 and _mm_cmpeq_epi64 were sse4 instruction,
there is not reason why using them, I re wrote using floats now, if you sync it soudl build now. 
```
DG_INLINE dgVector TestZero() const
{
	//return dgVector (_mm_cmpeq_epi32 (m_typeInt, m_zero.m_typeInt)) & m_negOne;
	return m_negOne & (*this == m_zero);
}
```

on he _POSIT__VER this is by design, it is added to the make file,  but I do not know if it is in the right place in the cmake script.  if you know a better place can you submit a patch?

leetNightshade, 2018-03-09:
Thanks for the fix! I guess I'll take the easy route and just predefine _POSIX_VER for now. I'm trying to setup a Genie build config, and trying to figure out best way to get around sse3 issues (buildoptions/cflag issue with genie/clang), so disabling that for now with DG_SCALAR_VECTOR_CLASS.

closed 2018-03-09 by leetNightshade