MADEAPPS/newton-dynamics #100: Ability to use your own game engine's threadpool

boberfly, 2018-07-09:
Hello!

I was just curious if this may or may not be planned, but will you implement a way from the API to set up your own job worker system instead of using Newton's thread hive system?

I was browsing through the source and sure enough you can just hack in your own system whenever there's a call to m_world->QueueJob most likely, or hack thread/hive code. My thoughts are more or less having tighter integration with a game engine which already supplies a threadpool, or perhaps run on a different thread model like with a fiber job system.

Cheers!

JulioJerez, 2018-07-13:
I do not have plan for that, but the thread poll class is very will insulated, you can add and if def and re implemented the way you'd like.
if the implementation is clean ad does no uses  code from private companies of skd SDK then I can merge it.
for example I can no use ps4 or xbox explicit code that can detect or enforce that a particular thread run on some specific core, which is what I think you mean by three fiber thong..

In any case I recently made a change the three thread pool to make so that each thread use his own queue, as oppose to before that each all work thread questioner jobs for a common pool which required
more atomic.
It turned out this method which with be good for you have control of thread priority, is really bad for 
c++ 11 thread model, because thread do no respond fast enough. 
so if for example you have two worker threads, and you assign and equal number of jobs to each thread,   
assuming on average each job has equal time, then the limiting factor is the last work tread that get activated.
if one other hand any thread start to execute jobs as far as they can , the even if the worker thread dare activate at same time, the will all complete at about the same time.
I will make the change to common queue soon.

It is too bad that the standard threads support system is designed to be PC centered an not for real time, 
but any end use with that can of expertise can make those changes.

boberfly, 2018-07-13:
Hi Julio thanks for the reply, I understand threading is a tricky thing to get right and I'm trying to understand it better myself as well.

I was thinking more or less a way to declare your own callback function in the API to your application's job/task model, or maybe something like the others are doing:
https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Threading.html#cpudispatcher

https://github.com/bulletphysics/bullet3/blob/master/examples/MultiThreading/MultiThreadingExample.cpp

I will have a look see, the fiber/job model I was going to see about enclosing all of Newton into a single job and all the calls to thread hive's QueueJob would spawn more jobs from that, not using PS4/Xbone but one of the open source libraries which just use like boost context under the hood for fibers and an MPMC queue to store the jobs to send to a threadpool eg. https://github.com/SergeyMakeev/TaskScheduler

Cheers Julio

JulioJerez, 2018-07-17:
few things is one one thing.
I will make some changes to the thread system. 
I will probably make so that is a a pointer similar to how the collision system is
the you can write your own thread pool and regretted a different of you like.

I remind you that I can't add any third party library to the engine, in particularly lib like boost which is extreme blooded and has their own licence.
Newton is a stand alone zlib library and I like to keep that way.
but once you can register a thread class then  you can keep it on your side.

nem0, 2018-10-15:
Good thing about the way e.g. PhysX does jobs is that the physics lib does not need to integrate any 3rd party. It just expose an interface that users implements, everything 3rd party is on user's side. It offers 100% flexibility and it's the only good way to really integrate physics in an MT game engine.

JulioJerez, 2018-10-15:
This is Newton is open source  feel free to change the thread system or use other engine. 
I am not investing any time on that.

closed 2019-03-31 by JulioJerez