MADEAPPS/newton-dynamics #101: assert is triggert when NewtonCreate is called

GenericNameClass, 2018-08-02:
The assert is called in File: dgCollisionSphere.cpp Line 156
Nothing else from newton is called before

Call Stack:
dgCollisionSphere::TessalateTriangle(...) Line 156
dgCollisionSphere::Init(...) Line 89
dgCollisionSphere::dgCollisionSphere(...) Line 44
dgWorld::dgWorld(...) Line 301
Newton::Newton(...) Line 41
NewtonCreate() Line 162

thanks in advance GenericNameClass

JulioJerez, 2018-08-02:
that should not happens p0 is supposed to be a unit vector.
dgAssert (dgAbs (p0.DotProduct3(p0) - dgFloat32 (1.0f)) < dgFloat32 (1.0e-4f));

look and the values on the watch window

GenericNameClass, 2018-08-03:
p0 is a unit vector

I think i found the problem, NewtonCreate is called from a constructor,which leads to that m_triplexMask
is not initialized at this moment,which leads to that p0.DotProduct3(p0) returns 0 which triggers the assert

JulioJerez, 2018-08-03:
Oh I see, are you using static links and you are making a global instance of a class the create the newton engine?
if so this is a problem of the CPP language that no one hos solve in the pass 40 years, mainly the initialization order of global variable is compiler vendor  dependent.
There are few things you can do to solve it. 
1-you can change the link order in the project file, but that is away trick it and to me is a hack, 
2- you can make your class a singleton, then you add a static function  like
static MyClass& getMyClass()
{  
   static  MyClass singleton();
  return singleton;
}

2- my preffer method is to no initialize anytgong in yor class and ass an init function   
class MyClass
{
   MyClass() 
      :world(NULL)
   {}

   void Init()
  {
      world = NewronCreate()
   }
   ....
   ... 
   Newton* world;
}

the you call Init is you Main function.
this I like better because does now realizes on compiles features.  and does not has the rigidity of singeltons

closed 2018-08-07 by GenericNameClass