MADEAPPS/newton-dynamics #102: reserved identifier violation

elfring, 2018-08-09:
I would like to point out that identifiers like “[`__NEWTON_H__`](https://github.com/MADEAPPS/newton-dynamics/blob/3c73e735541d95ca9d3cb8d07db3120d1a045790/sdk/dgNewton/Newton.h#L22 "Update candidate")” and “[`__DG_PROFILER_H__`](https://github.com/MADEAPPS/newton-dynamics/blob/eaef1baa67ac944893de8ef6994b3b808dfafe82/sdk/dgCore/dgProfiler.h#L22 "Another update candidate")” [do not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51-CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

JulioJerez, 2019-03-31:
I do not consider thsi a serius issue. Try to imposue language rule to sime write before thsio rule were created does not make sence.

closed 2019-03-31 by JulioJerez

elfring, 2019-03-31:
How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

JulioJerez, 2019-03-31:
what do you mean undefined behavior
what is wrong with this?
#ifndef __NEWTON_H__
#define __NEWTON_H__

elfring, 2019-03-31:
I suggest to [reconsider the usage of (double) underscores in affected identifiers](https://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier#answer-228797 "What are the rules about using an underscore in a C++ identifier?").

JulioJerez, 2019-03-31:
but the define use double underscore, this is how si define, 
#ifndef __NEWTON_H__
#define __NEWTON_H__

but what I do not understand is what difference does it make use or not using undeercore?

elfring, 2019-03-31:
Please [reduce the tampering with the reserved name space](https://en.cppreference.com/w/c/language/identifier#Reserved_identifiers "Description for reserved identifiers").
Would you like to achieve compliance with known programming standards?

JulioJerez, 2019-03-31:
what standard? 
If someone made a starndard about he or she like defines to be declare, them they can use it on thier own stuff. no one is obligated to use Newton.
I am not changing that, this is my final answer.

elfring, 2019-03-31:
Please take another look at the rules for the usage of reserved identifiers in the C++ programming language.