MADEAPPS/newton-dynamics #103: Remove unnecessary null pointer checks

elfring, 2018-08-09:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [dGeometryNodeSkinModifierInfo](https://github.com/MADEAPPS/newton-dynamics/blob/d80d3459c25283e63904c98bfab08d0f2f83d6d9/sdk/dScene/dGeometryNodeSkinModifierInfo.cpp#L69 "Destructor for dGeometryNodeSkinModifierInfo")
- [dNewtonJoint::Destroy](https://github.com/MADEAPPS/newton-dynamics/blob/222ad40b98ee44c9cdf1c72afffec9de6ebf69a4/sdk/dWrappers/csharp/wrapperSdk/dNewtonJoint.cpp#L42 "dNewtonJoint::Destroy function")
- [WRAP_NewtonDestroyCustomJoint](https://github.com/MADEAPPS/newton-dynamics/blob/222ad40b98ee44c9cdf1c72afffec9de6ebf69a4/sdk/dWrappers/csharp/CustomJoints.cpp#L11 "WRAP_NewtonDestroyCustomJoint function")

elfring, 2018-08-20:
>…, but since the engine uses his own pool base allocators, this is not true for the Newton sdk.

How can this be?

If a customised C++ delete operation was safely implemented, it should tolerate the passing of null pointers, shouldn't it?

>This is in fact part of my programming style.

Would you like to reconsider an aspect in the coding style?

JulioJerez, 2018-08-20:
that easy, and quite common, basically you have a class with an object that can be destroyed.
but is more general, a collision of bodies and joint form a general undirected graph when there can be island that form circular dependencies, for the a more complex mechanism for handling dependence is needed. for example say a body is part of some connected articulation of joint an bodies.
the application may have a  reference to one or more joint to control the articalation.
if a body is deleted, the engine has to destroy the incident joint to that body.
this is trivial in languages like C shap and Java, but not in C++ so the engine has to have a memory manage that take care of that, he is the function call back that is call when some like that happens, you can see that the joint is deleted but no ta class. the at some later point the class is deleted, but then that class will have some members that were destroyed already.

```
void dCustomJoint::Destructor (const NewtonJoint* me)
{
	// get the pointer to the joint class
	dCustomJoint* const joint = (dCustomJoint*) NewtonJointGetUserData (me);  

	joint->m_autoDestroy = 1;

	// delete the joint class
	delete joint;
}

```

>  Would you like to reconsider an aspect in the coding style?

well no, I happen to disagree wit that CPP rule.

elfring, 2018-08-20:
* It seems that you are concerned for referential integrity to some degree.
* I propose just the deletion of a few redundant condition checks for this issue.
  Would you like to increase the usage of smart pointers also for this software instead?

JulioJerez, 2018-08-20:
but why do you want that to be changed when it is not broken, newton is a very low level library it does not required smart pointers, smart pointers are in fact a very naïve programing paradigm that only handle acyclic graphs. not circular depenedies.  

you may not see it, but the scene manager is the engine is quite sophisticated, it handle the scene far more efficiently that languages with native garbage collision support and does not requires garbage collision passes.

elfring, 2018-08-20:
I would find a small source code reduction nice because of less questionable pointer checks.

elfring, 2018-08-20:
The deletion of redundant pointer checks can be achieved with incremental software update steps, can't it?

JulioJerez, 2018-08-20:
It can, but we do not have to,  a deleted object is deleted immidially, not garbage collection and no reference counting needed. 
people can use those thong on top, but the low lever do just fine.
this is such a picky issue that I do not know why is so important to your,  it does not violates any rule of the language.

elfring, 2018-08-20:
* Can extra null pointer checks (before calls of the C++ delete operator for example) reduce the desired software efficiency?
* Is any code nicer when you do not have to write it yourself (because it could be provided by a known standard library)?

elfring, 2018-08-21:
Can it be that functions like the following belong also to a software category which [should tolerate the passing of null pointers](https://wiki.sei.cmu.edu/confluence/display/c/EXP34-C.+Do+not+dereference+null+pointers "Do not dereference null pointers.")?
* [dgFreeStack](https://github.com/MADEAPPS/newton-dynamics/blob/99644d06aee1b5c4558584f825a9cd0735be8b09/sdk/dgCore/dgMemory.cpp#L399)
* [dgMemFree](https://github.com/MADEAPPS/newton-dynamics/blob/99644d06aee1b5c4558584f825a9cd0735be8b09/sdk/dgCore/dgMemory.h#L37)
* [dgMemoryAllocator::FreeLow](https://github.com/MADEAPPS/newton-dynamics/blob/99644d06aee1b5c4558584f825a9cd0735be8b09/sdk/dgCore/dgMemory.cpp#L217)
* [NewtonFree](https://github.com/MADEAPPS/newton-dynamics/blob/33029a10dde21c0049dd99277a40b4429cf93965/sdk/dgNewton/Newton.cpp#L135)

JulioJerez, 2018-08-21:
no it does no reduce efficiency, you are checking for null, on the destructor, as opposed to using smart pointer which are externally poor in every regard.

you keep asking the same question on different classes, I already told you this is my style and that I disagree with that Cpp rule that allows to call delete on Null.

elfring, 2018-08-21:
I find the disagreement for a “Cpp rule” questionable.
I assume that it is derived from very specific software development concerns.

Smart pointers can not be that bad (according to your impression) when [they belong to current programming recommendations](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "C++ core guideline “R.11: Avoid calling new and delete explicitly”").
How are the chances to improve affected destructor implementations generally?

:crystal_ball: Can it be nice to achieve null pointer tolerance (or [avoidance of undefined software behaviour](https://wiki.sei.cmu.edu/confluence/display/c/EXP34-C.+Do+not+dereference+null+pointers "Do not dereference null pointers.")) by corresponding checks at central source code places?

JulioJerez, 2018-08-21:
well I am sorry you are so strict with those standard.
I do not think I Am in a minority when I say I disagree with some of those cpp standarts 
but so be it.

there will be not call to function with null pointers in the newton sdk, if this is a show stopper for you. The only thing I can say is the perhaps some other open source engine is a better fix for you.

elfring, 2018-08-21:
>… I say I disagree with some of those cpp standarts …

You are free to express your own opinions around such topics.

:thinking: But I guess that we can agree on the kind of source code that should usually be put into constructors and destructors of classes in a programming language like “C++”.

>there will be not call to function with null pointers in the newton sdk, …

It is nice when this will be usually the case.

But some (C++) programming interfaces require to handle null pointers in reasonable ways. After you know that some software implementations take care of such a development concern, I do not see a need to repeat similar actions at other source code places.

Some checks can be moved into additional constructor/destructor combinations (provided by class attributes).

closed 2019-03-31 by JulioJerez