MADEAPPS/newton-dynamics #104: Completion of error handling

elfring, 2018-08-09:
Would you like to add more error handling for return values from functions like the following?
- [fopen](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fopen.html "Open a stream") ⇒ [ShaderPrograms::LoadShaderCode](https://github.com/MADEAPPS/newton-dynamics/blob/1f368ee67912dfc41b9dfa8f36e93387ecb8cd95/applications/modelEditor/nme/toolBox/ShaderPrograms.cpp#L59)
- [fclose](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fclose.html "Close a stream") ⇒ [dLexScannerGenerator::SaveFile](https://github.com/MADEAPPS/newton-dynamics/blob/e5a847f80ffc0310db309e616fb6d357f82a7e7e/sdk/dCompilerKit/dLexicalGenerator/dLexScannerGenerator.cpp#L176)

JulioJerez, 2018-08-20:
that functionality is a legacy from newton 1.5 that is not used anymore.
the code was removed and now deleted since it cause problem for the automatic cmake script.
newton does not try to make stride on graphics, it simply uses the bare minimum which provide d by open go basics.

elfring, 2018-08-20:
Can used software components be better identified anyhow currently?

JulioJerez, 2018-08-20:
can you rephrase that question, I do not understand?

elfring, 2018-08-20:
How do you think about to improve [static source code analysis](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2C_C.2B.2B "Tools for code analysis") also for this software?

-----

* Are you interested to apply [aspect-oriented software development](https://aspectc.org/ "Software development with AspectC++?")?
* Would you like to encapsulate error detection and corresponding exception handling as [a reusable aspect](https://en.wikipedia.org/wiki/Aspect_%28computer_programming%29 "Combination of a pointcut with advice") in this software?

closed 2019-03-31 by JulioJerez