MADEAPPS/newton-dynamics #106: Wiki links are broken

HiceS, 2018-08-15:
Trying to access the wiki links for building because of CMAKE issues and noticed the wiki links have been broken.

JulioJerez, 2019-03-31:
CMake was re factore and nwo work for PC, Linux and Macs

closed 2019-03-31 by JulioJerez

JayFoxRox, 2019-04-12:
Issue should not have been closed - the GitHub Wiki links are still broken: The header correctly directs to Mediawiki, but none of the other links on GitHub are valid as the content wasn't migrated yet.

*(The note about CMake was unrelated and only explained why the Wiki issue was noticed)*

JulioJerez, 2019-04-12:
well, I do not maintain the wiki,  it was a community project that started about 10 years ago, 
since them people has abandoned and move on..