MADEAPPS/newton-dynamics #108: forum mysql permissions

JernejL, 2018-08-19:
Something went wrong with the hosting again: 
 
> General Error
> SQL ERROR [ mysql4 ]
> 
> UPDATE command denied to user 'ngdforum'@'97.74.24.133' for table 'phpbb_sessions_keys' [1142]
> 
> An sql error occurred while fetching this page. Please contact an administrator if this problem persists.
 
You need to configuire  'ngdforum'@'97.74.24.133' user to have select, insert, update permission on ngdforum.* tables, can you do this in your hosting control panel?

JulioJerez, 2018-08-19:
yes giddy sent me an email tell me that the forum database is 0.02 gnytes over the limit.
I called them and they say it was a false alarm, but that they could not fix the forum software, that I need to get someone to fix it or that I can use mySql to fix it myself, but that stuff is gibberish to me.
can you do that Delfi?

I will send you the email

JernejL, 2018-08-20:
You can email me at jernejcoder@gmail.com , however i think what they did was remove permissions for the user forum used because they thought we were over max size and maybe the problem is simple to solve: 
 
If you can get to phpmyadmin inside cpanel or something like that you can run following query to give it permissions again, or ask them to set these permissions: 
 
GRANT SELECT, INSERT, UPDATE, DELETE, ALTER ON ngdforum.* TO 'ngdforum'@'97.74.24.133'; 
 
If possible, can you add me as technical contact for the hosting? i could help with this, the host seems to break something every few months :)

JulioJerez, 2018-08-20:
I forwarded you the email they sand me.  This is the account name and password

 

go daddy username:  juliojerez

password: 4475Jrj#7

JernejL, 2018-08-20:
Julio i think you should delete that and change password on godaddy immediately. 
 
I assume you wanted to send that via email, but github had linked the reply to the comment thread..

JulioJerez, 2018-08-20:
oh sorry about tha mistake, I deleted the post

TrevorCash, 2018-08-20:
forums are still down for me as well

svenberra, 2018-08-24:
Any news on this? I'm starting to get physics forum abstinence. O.o
//Sweenie

JernejL, 2018-08-24:
Still nothing new, i've trimmed database for fat i could remove, but godaddy hasn't re-enabled write mode on database yet

JulioJerez, 2018-08-24:
I just called Go Daddy again, and they said that yes the database is about 75 mg, that the system should re enable automatically,  but that can take for 24 to 72 hours to take place.
he said it has not been 72 hours yet for the to do it manually.
we just need to wait one more day.

JernejL, 2018-08-27:
Please do call them, they told me via chat just to call them which i can't. 
I could not find any option to correct / re-enable database write on their management portal. 
 
They should have restored the database permissions already, this is very incompetent from godaddy :\

JulioJerez, 2018-08-27:
well this is seven days already, I will call again tomorrow, and see what they have to say. we already comply we everything they requested, they will have to do it manually.

JulioJerez, 2018-08-27:
I got this email

Discussion Notes
--
Support Staff Response
Dear Sir/Madam,In regards to your hosting account forÂ newtondynamics.com:Our administrators were able to correct an internal configuration causing the issue you were experiencing. Your issue is resolved at this time.We have decided to no longer offer general email support to ensure the best possible experience for you and our other customers. So if you have any follow-up questions please reach out to our other support teams here:http://support.godaddy.com/You may still include this incident ID for reference when contacting our Customer Care teams. It is important to us to fully resolve your issue and we look forward to assisting you.Our service department is available 24 hours a day, 7 days a week, throughout the year to assist you at any time. You can contact one of our representatives by calling 1- (480) 505-8877.Regards,Dennis V.Hosting Support

JernejL, 2018-08-28:
Yup, they have solved it, the forums function correctly now!