MADEAPPS/newton-dynamics #109: Improve exception safety with smart pointers

elfring, 2018-08-21:
Would you like to [wrap any pointer data members](https://github.com/MADEAPPS/newton-dynamics/blob/de0ac32822a83d735627d4f7f7962aa0448ea3ae/sdk/dNewton/dMaterialPairManager.cpp#L16 "Update candidate: dMaterialPairManager class") with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

JulioJerez, 2019-03-31:
Newton is not a  CPP 11 library, tryning adding C++11 feature is a big mistake.

closed 2019-03-31 by JulioJerez

elfring, 2019-03-31:
Do you care for corresponding software correctness?

JulioJerez, 2019-03-31:
who said the code is incorrect?  
code has been written before the inversion of std::unique_ptr
if this is a show stopper for you, I am sorry but I am no spending time of that.

elfring, 2019-03-31:
Source code which contains multiple delete statements in destructors is questionable today.
How do you think about [the C++ guideline “R.11: Avoid calling new and delete explicitly”](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "Recommendations around allocation and deallocation")?

JulioJerez, 2019-03-31:
liek I said before, newton is no a cpp 11 library.

elfring, 2019-03-31:
This view does not really matter for an implementation detail which is questionable also in this software.
Do you understand other information sources better for the explanation of this open issue?

JulioJerez, 2019-03-31:
perhaps you will better served using some other library more conformance to your liking.
I am not making these changes, that my final answer of this.