MADEAPPS/newton-dynamics #110: Fix destructor for dBaseHierarchy

elfring, 2018-08-21:
I find [the specification of the following loop](https://github.com/MADEAPPS/newton-dynamics/blob/de0ac32822a83d735627d4f7f7962aa0448ea3ae/sdk/dContainers/dBaseHierarchy.cpp#L27 "Update candidate: Destructor for dBaseHierarchy") inappropriate.

```C++
…
	while (m_child) {
		delete m_child;
	}
…
```

* Will such source code result in an endless loop if the used variable will contain a pointer which is not null?
* How do you think about to [avoid multiple delete operator calls for the same object](https://wiki.sei.cmu.edu/confluence/display/cplusplus/MEM51-CPP.+Properly+deallocate+dynamically+allocated+resources#MEM51-CPP.Properlydeallocatedynamicallyallocatedresources-NoncompliantCodeExample%28Double-Free%29 "Properly deallocate dynamically allocated resources.")?

JulioJerez, 2018-08-22:
> I find the specification of the following loop <https://github.com/MADEAPPS/newton-dynamics/blob/de0ac32822a83d735627d4f7f7962aa0448ea3ae/sdk/dContainers/dBaseHierarchy.cpp#L27>  inappropriate.
>
> …

no it won't, the destructor of the child changes the pointer, this ha been working for more than 20 years.

elfring, 2018-08-22:
>…, the destructor of the child changes the pointer, …

I found it not so obvious that a pointer modification can happen there.

Can the following statement be also sufficient (without a loop)?

```C++
…
	delete m_child;
…
```

Can a corresponding smart pointer take care of the desired clean-up instead?

JulioJerez, 2018-08-22:
no, imagine you have an intrusive link list where the pointer to the next element on the list is stored on each element.
to delete a node and all its elements you  simple delete the first element, 
the destructor unlink the node,  set the next pointer to the next next and delete next in a while loop until it reach the last.

this is done extensively on all template classes of the library.
you can set a break point and see how it works. not recursion required.

elfring, 2018-08-22:
Thanks for your explanation of the “intrusive” data structure design approach at this place.

closed 2018-08-22 by elfring

elfring, 2018-08-23:
:thinking: Which source file does contain the operator delete implementation which modifies the pointer in the mentioned variable?