MADEAPPS/newton-dynamics #112: Bunch of errors while building "no LIBRARY DESTINATION for shared library target"

marchevskys, 2018-08-29:
log from CMake:
Steps:
> -- mkdir build
> -- cd build
> -- cmake ..
What is wrong?

> -- The C compiler identification is GNU 7.3.0
> -- The CXX compiler identification is GNU 7.3.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> dMath
> dgCore
> dScene
> CMake Error at sdk/dScene/CMakeLists.txt:25 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dScene".
> 
> 
> CMake Error at sdk/dScene/CMakeLists.txt:47 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dScene".
> 
> 
> dNewton
> CMake Error at sdk/dNewton/CMakeLists.txt:24 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dNewton".
> 
> 
> CMake Error at sdk/dNewton/CMakeLists.txt:46 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dNewton".
> 
> 
> newton
> CMake Error at sdk/dgNewton/CMakeLists.txt:24 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "newton".
> 
> 
> CMake Error at sdk/dgNewton/CMakeLists.txt:43 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "newton".
> 
> 
> dgPhysics
> dTimeTracker
> CMake Error at sdk/dProfiler/dTimeTracker/CMakeLists.txt:31 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dTimeTracker".
> 
> 
> CMake Error at sdk/dProfiler/dTimeTracker/CMakeLists.txt:32 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dTimeTracker".
> 
> 
> dAnimation
> dContainers
> CMake Error at sdk/dContainers/CMakeLists.txt:24 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dContainers".
> 
> 
> CMake Error at sdk/dContainers/CMakeLists.txt:44 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dContainers".
> 
> 
> dCustomJoints
> CMake Error at sdk/dCustomJoints/CMakeLists.txt:24 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dCustomJoints".
> 
> 
> CMake Error at sdk/dCustomJoints/CMakeLists.txt:45 (install):
>   install TARGETS given no LIBRARY DESTINATION for shared library target
>   "dCustomJoints".
> 
> 
> dgNewtonAvx
> tinyxml
> glfw
> imgui
> demosSandbox
> DebugRelease
> -- Configuring incomplete, errors occurred!

marchevskys, 2018-08-30:
I would like to fix it myself, to make it usable for Linux. Need your help :)

arpio, 2018-10-24:
Hi Julio. With the files from here http://newtondynamics.com/forum/viewtopic.php?f=9&t=9282 and some additions my clone https://github.com/arpio/newton-dynamics is now building on Linux with Demos(but not all demos are working).

If you not to busy can you check the changes in the source files? Only with this changes I could build the demos on Linux.

JulioJerez, 2018-10-24:
oh nice, can you post in the forum and remind me Saturday, so that I go over and merge the changes?

JulioJerez, 2019-03-31:
fixed

closed 2019-03-31 by JulioJerez