MADEAPPS/newton-dynamics #117: DG_MEMORY_GRANULARITY_BITS detection is wrong and can lead issues on 64bit platform (crash memory).

execomrt, 2018-09-18:
Please use this macro using __LP64__. At worst, it would lead 64bit alignment.

	#if !((defined (_WIN_64_VER) || defined (_MINGW_64_VER) || defined (_POSIX_VER_64) || defined (__LP64__)))
		#define DG_MEMORY_GRANULARITY_BITS		5
	#else
		#define DG_MEMORY_GRANULARITY_BITS		6	
	#endif

JulioJerez, 2018-09-18:
why would that lead to crash in 64 bit systems?
it has worked for 15 years without problems.
that define has nothing to do with memory alignment or anything like that.
is has to do with chunk information on newton internal chunk system,
basically in requires a 32 bytes header for saving internal info and that
size double with 64 bit pointers.

JulioJerez, 2018-09-18:
oh I see what you mean, LP64 is a defined that maybe be more general.
I will added yes.

JulioJerez, 2018-09-18:
another thing I can do is to just use the 64 bit even on 32 build mode since
32 bit is are on their wait out.

execomrt, 2018-09-18:
Yes also I can show you a simple project on Android running on arm64. I needs this to be specified else the dgMemoryAllocator get crazy result(dgMemory.cpp line 261, charPtr - paddedsize)

closed 2019-03-31 by JulioJerez