MADEAPPS/newton-dynamics #120: dgTypes.h : dgScopeSpinPause uses explicitly  _mm_pause(), it should be only for x86 platform.

execomrt, 2018-09-18:
Should have
#if defined (__i386___) || defined (__x86_64___)
_mm_pause();
#endif

Note that Android supports x86, but most of the source check for WIN32 instead : This requires a lot of refactor.

execomrt, 2018-09-19:
I suggest std::this_thread::yield() for other platform (C+11 is now mature everywhere)
https://en.cppreference.com/w/cpp/thread/yield

(Also std::atomic<int> value for all your atomic increments).

JulioJerez, 2018-09-19:
> (Also std::atomic value for all your atomic increments).

it is not a thread yield, it is not meant to relequich the thread time, it
just need to prevent other threads from writing to the same location at the
same time.
basically is used as an atomic write.
if using yield, it does a task switch and is much slower.
the mm_pause, according to intel flush the cup pipe forcing to do the
memory read each time.

execomrt, 2018-09-19:
I know, but we need 'something' when running on non x86 (ARM)

Or move everything to std::memory_order_acquire()

https://en.cppreference.com/w/cpp/atomic/memory_order

execomrt, 2018-09-19:
Here an example of the 'ScopeLock' rewritten (I've kept the timeout).
```
#include <atomic>

class ScopeLock
{
public:
	inline ScopeLock(std::atomic<bool>& lock) : m_lock(lock) {
		const int maxCount = 1024 * 32;
		for (int i = 0; (i < maxCount) && m_lock.exchange(true, std::memory_order_relaxed); i++) {
			std::atomic_thread_fence(std::memory_order_acquire);
		}		
	}
	inline ~ScopeLock()
	{
		std::atomic_thread_fence(std::memory_order_release);
		m_lock.store(false, std::memory_order_relaxed);
	}

	std::atomic<bool>& m_lock;
};

and replace all your unsigned m_lock; 
to std::atomic<bool> m_lock = {false}

```

JulioJerez, 2018-09-19:
I do not think I can riddle the high level code with c++11 types, there are users that still build the engine with vs 2010 and 2013.
the loop does not even have to do the mm_pause on non intel cpus.
on amds mm_pause resolve to nop.
I also have tested spin lock like that on ARM system and doing nothing seem to work fine.

whe can probably make it for x86 use mm_pause and for other the standart thread yield. the mm_pause becomes and issue on intel cup that if the loop is so small the it gets all cached, them the loop never read the atomicfrom memory because is it cache and the branch is always predicted.
inserting a bunch of nop would the same but intel made that intrinsic just for that purpose.

execomrt, 2018-09-19:
Atomics are supported since VS 2012. A quick check for MSC_VER would do the trick.

Doubt that some people still working on VS 2010 in 2018, considering that VS 2015/2017 are free (Community version).

JulioJerez, 2019-03-31:
and yet ther are still system that are build in VS2010

closed 2019-03-31 by JulioJerez