MADEAPPS/newton-dynamics #123: ConvexHull collision generation is offset

TrevorCash, 2018-09-21:
ConvexHull collision generation is offset incorrectly.

This is using:
NewtonCreateConvexHullFromMesh

NewtonCreateCompoundCollisionFromMesh seems to work fine and is seen on the right hand side of the image:

[image]

TrevorCash, 2018-09-21:
when forming the newton mesh I have no added offset.  so all vertices are centered around 0,0,0

I use the same newton mesh for both procedures. 

I added an explicit 
`NewtonCollisionSetMatrix(newtonCollision_, &dGetIdentityMatrix()[0][0]);`

After forming each collision to make sure I am getting an identity transform before I pass it onto my system.  But the offset is still the same.

Making the compound:
```
        newtonCollision_ = NewtonCreateCompoundCollisionFromMesh(world, newtonMesh_->mesh, hullTolerance_, 0, 0);

        NewtonCollisionSetMatrix(newtonCollision_, &dGetIdentityMatrix()[0][0]);
```

Making the Convex Hull:

```
        newtonCollision_ = NewtonCreateConvexHullFromMesh(world, newtonMesh_->mesh, hullTolerance_, 0);

        NewtonCollisionSetMatrix(newtonCollision_, &dGetIdentityMatrix()[0][0]);

```

TrevorCash, 2018-09-21:
If I corrected myself would the offset be something like  0.5*(mesh bounding height) ?

of course I'd love to avoid that if possible - keep things clean.

JulioJerez, 2018-09-21:
I presume you mesh has a pivot flush wi the lower end, as many game assets do.
you should not have to do anything at all, the function should make a collision shape that matched the  visual mesh.
I just went to demo UsingNewtonMeshTool.cpp and offset the box point so that the pivot if flush with floor, can you see if you can adapt the demo to reproduce that error? 
that demo is very simuilar to what you are doing.

TrevorCash, 2018-09-21:
I looked at the demo and as an experiment I shifted the coordinates of all vertices in the sample box by 1 in all 3 directions.  

I also changed the second box so that it uses NewtonCreateCompoundCollisionFromMesh.

If you turn on both show center of mass and show body frame you can see the difference in the offset of internal collision on the collision shapes.

See this commit: https://github.com/TrevorCash/newton-dynamics/tree/DemoMeshTool_CompoundHullvsHull

TrevorCash, 2018-09-21:
Also if you comment out the "shifting by one" part in both CreateSimpleBox_NewtonMesh and CreateSimpleBox_dNewtonMesh you will see no offset from center of mass to body frame on both bodies.

TrevorCash, 2018-09-21:
I was able to get my app working by utilizing the following code if it helps:
```
        newtonCollision_ = NewtonCreateConvexHullFromMesh(world, newtonMesh_->mesh, hullTolerance_, 0);

        NewtonCollisionSetMatrix(newtonCollision_, &dGetIdentityMatrix()[0][0]);

        //offset by model geometry center (#todo this may be a quirk of newton - CompoundHulling does not need this.)
        position_ += model_->GetGeometryCenter(modelLodLevel_);
```

JulioJerez, 2018-09-21:
that's what I was telling you in my first post. 
The convex will have a no identity matrix because is calculate the hull relative to the geometric center of the point cloud.

what I find weird is that your mesh must have some offset matrix use for rendering, 
your vertices are not centered around the origin as you.

TrevorCash, 2018-09-21:
I see.  I guess the only thing that threw me off is that it behalves different with NewtonCreateCompoundCollisionFromMesh.

Urho3D has an offset for models yes.  By default it computes it as the center of the bounds. So when I render, models will always render equally around the center of the model.  The data I feed to newton using the mesh utility however has no offset.  when forming the body - I pass the collision right in, with an optional user defined offset (position_).

closed 2018-09-21 by TrevorCash