MADEAPPS/newton-dynamics #130: max_DOF assert with lots of objects

TrevorCash, 2018-09-28:
I have synced with the latest code - I am getting a dgAssert(contact->m_maxDOF); when I spawn a lot of free objects.  I have no joints in my test scene just bodies with single collision shapes falling against a terrain.

[image]

In release mode this actually results in a crash I think.

TrevorCash, 2018-09-28:
Ok, yeah it's pretty repeat able after I spawn a few hundred objects or so

JulioJerez, 2018-09-28:
ok I think the bug is just a typo,  and active joint is one that has both contact and is no sleeping.
this line 
			if (contact->m_contactActive || contact->m_maxDOF) {
should be like this
                       if (contact->m_contactActive && contact->m_maxDOF) {

is fix now please check again

TrevorCash, 2018-09-29:
Ok Yeah That fixed that.  I am able to spawn alot more objects though.

How ever now I see that I am getting a different assert after I spawn quite alot (probly 1000 or so):
[image]
[image]

JulioJerez, 2018-09-29:
ok, yes that one is also a warning I added to verify that the new algorithm that build island actually run in linear time proportion to the number of bodies.  
can you set to m_rank <= 6  temporarilly only I make that change. 

I think is has been tested long enoght that we can assume that on average the number of test will be less than 6

Is good you test in debug mode, but also notice these assert are compiled out in release mode.

TrevorCash, 2018-09-29:
Ok I'll leave that one in there as it is.

JulioJerez, 2018-09-29:
the rank is upped by one now, keep those bugs coming, it will only make the engine stronger.

closed 2018-10-01 by TrevorCash