MADEAPPS/newton-dynamics #142: Can't compile

dustContributor, 2019-01-13:
Hi! I'm trying to compile Newton as a shared lib on Debian x64, using gcc 8.2

I'm getting these errors:

```
In file included from /newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.cpp:13:
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h: In member function ‘int dAnimTakeData::dAnimTakeTrack::dAnimTakeArray<OBJECT>::GetIndex(float) const’:
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:50:22: error: there are no arguments to ‘GetSize’ that depend on a template parameter, so a declaration of ‘GetSize’ must be available [-fpermissive]
     const int size = GetSize();
                      ^~~~~~~
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:50:22: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:51:13: error: ‘m_data’ was not declared in this scope
     if (t > m_data[size - 1].m_time) {
             ^~~~~~
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:51:13: note: suggested alternative: ‘__dt ’
     if (t > m_data[size - 1].m_time) {
             ^~~~~~
             __dt 
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:61:14: error: ‘m_data’ was not declared in this scope
      if (t < m_data[mid].m_time) {
              ^~~~~~
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:61:14: note: suggested alternative: ‘__dt ’
      if (t < m_data[mid].m_time) {
              ^~~~~~
              __dt 
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:69:10: error: ‘m_data’ was not declared in this scope
      if (m_data[i].m_time >= t) {
          ^~~~~~
/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:69:10: note: suggested alternative: ‘__dt ’
      if (m_data[i].m_time >= t) {
          ^~~~~~
          __dt 

```

dustContributor, 2019-01-13:
I understand you are working on these files right now, if this is to be considered unstable, could you point me to a commit where the library is considered stable?

JulioJerez, 2019-01-14:
you are right I am getting theses errors
In file included from /home/julio/Desktop/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.cpp:13:0:
/home/julio/Desktop/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h: In member function ‘int dAnimTakeData::dAnimTakeTrack::dAnimTakeArray<OBJECT>::GetIndex(float) const’:
/home/julio/Desktop/newton-dynamics/sdk/dAnimation/dAnimIKBlendNodeTake.h:49:22: error: there are no arguments to ‘GetSize’ that depend on a template parameter, so a declaration of ‘GetSize’ must be available [-fpermissive]
     const int size = GetSize();

I am not sure why. as far as I know that class is correct, there are about a dozen other template classes like that in the sdk and they all work, as far as I know thsi is a GCC bug.

If anyone can figure out the solution I accept a pull request, but now I have not idea why does not compile with GCC
                      ^~~~~~~

JulioJerez, 2019-01-14:
Ah I fid out hwo to get it working, this oen of those pedante GCC thing, apparently  in thsi template function
```
			int GetIndex(dFloat t) const
			{
				dAssert(t >= 0.0f);
				const int size = GetSize();
```
GCC can not figure out how to resolve the name decoration of any base class function and I have to explicitly call for the operation this, like this. 

```
			int GetIndex(dFloat t) const
			{
				dAssert(t >= 0.0f);
				const int size = this->GetSize();
```

I have only seen this with GCC, this does no happen with Clang or any version of visual studio.
by anyway it is fixed now, please sync again.

dustContributor, 2019-01-20:
Perfect, now it works, thanks! :smile: 

I have one unrelated question, do you support compiling everything into a single .so/.dll file? Building without the demos and such I get a bunch of libdgCore.so, libdMath.so, libdNewton.so and so on. Would it be possible to compile everything into a single .so/.dll without redoing the CMakeLists.txt scripts?

JulioJerez, 2019-01-20:
you do not have to make then DLLs you can make them static libraries, and link to your executable.
There an utility that can appen static librarye together in one archive.

There is also a way to combine multiple static libraries into one shared library. In GNU the command is like this
g++ -shared -o oneLibrary.so -Wl,--whole-archive libOne.a libTwo.a  … 

In visual studio is a command in the IDE that merge all the .lib together
this was how was done in the direct makefile in the 1.5 before going open source.

dustContributor, 2019-01-20:
> There is also a way to combine multiple static libraries into one shared library in GNU, the command is like this: g++ -shared -o oneLibrary.so -Wl,--whole-archive libOne.a libTwo.a …

Worked like a charm.

```
gcc -shared -o newton.so -Wl,--whole-archive libdAnimation.a libdContainers.a libdCustomJoints.a libdgCore.a libdgPhysics.a libdMath.a libdNewton.a libdScene.a libdVehicle.a libnewton.a libtinyxml.a -Wl,--no-whole-archive
```

Thanks!

closed 2019-01-20 by dustContributor