MADEAPPS/newton-dynamics #143: demosSandbox cannot be run directly from visual studio without first manually copying media assets to build directory.

TrevorCash, 2019-02-15:
using cmake to build vs solution:

demosSandbox cannot be run directly from visual studio without first manually copying media assets to build directory.

DaveGravel, 2019-02-15:
I think it happen because with cmake you don't use the default build folder.

TrevorCash, 2019-02-15:
Ah - yes that makes sense.  although It might be nice to get cmake to copy the assets in the case of isolated build folder?

JulioJerez, 2019-02-15:
yes I notice that? how is that done?

DaveGravel, 2019-02-15:
I never try but maybe it can help : https://stackoverflow.com/questions/13429656/how-to-copy-contents-of-a-directory-into-build-directory-after-make-with-cmake

d3x0r, 2019-03-09:
or - just runthe install target, change where you are debugging,

JulioJerez, 2019-03-31:
Thsi was not by design, but afte see the opration It is a desired funtionality.
so it is a feature.

closed 2019-03-31 by JulioJerez