MADEAPPS/newton-dynamics #146: Crash in dgBody::GetFirstContact

TrevorCash, 2019-02-15:
Got a random crash where joint is null in dgBody::GetFirstContact

[image]

Not yet sure how to reproduce

JulioJerez, 2019-02-16:
can you set a breakpoint on the function and run in debug to see the trace stack?
I do not think the trace stack is valid.

TrevorCash, 2019-02-16:
this is a better stack trace in my usage:
[image]

my function is formContacts() which iterates over all contacts of a body.  somehow node->GetInfo().m_joint is null - even though the body is valid in dgBody::GetFirstContact()

TrevorCash, 2019-02-16:
Here is the call in my code that crashes: 
[image]

JulioJerez, 2019-02-16:
tell tow more things.
1-Is thsi new wit the double, or does I happens in single mode
2-does this happen in single threaded mode.
3-in the trace stack I do not see any link for a newton call back. are you doing it from outside a newtion update?

I have no test that for a while, and with teh changes is possible is bloke, It seem impossible that a node in teh graph has a null contact, but that what is seem to be.
if you can answer those question so that I can set a debug case.

JulioJerez, 2019-02-16:
Ah also I assume this was a debug build?

TrevorCash, 2019-02-16:
Yes Debug, multithreaded,

I am not using the callback - I am iterating the contacts outside the update

JulioJerez, 2019-02-16:
Ah iterating outside the update should be even more secure.
Just for narrowing some more, can you try in single thread mode?

TrevorCash, 2019-02-17:
I can't get it to crash in single threaded.

Here is the Debug, multithreaded demos - they are hard for me to get to crash now, So I'm not sure.  maybe a script could be made to loop-start the program untill it crashes.

https://drive.google.com/open?id=1NBjk9PZjWfUiLgbkGrRJB7gFTYacmgwn

TrevorCash, 2019-02-17:
the 11_physics should in theory give the original crash.
201_physicstests gives aanother repeatable crash I am not sure what is causing.

JulioJerez, 2019-02-17:
I see the assert, but I can't debug static libraries. you to linked it to a newton dll, 

can you link to the newton dlls?
also tell me what dll, double or single, 64 or 32 bit?

JulioJerez, 2019-02-17:
It seem some routine is generation a bogus contact point with  a bad normal here
DG_INLINE dgVector dgCollisionInstance::SupportVertexSpecial (const dgVector& dir, dgInt32* const vertexIndex) const
{
	dgAssert (dir.m_w == dgFloat32 (0.0f));
	dgAssert(dgAbs(dir.DotProduct(dir).GetScalar() - dgFloat32(1.0f)) < dgFloat32(1.0e-2f));

the may be a bug in teh double vector class, are you linking to double?
al I did not see the plugin dlls, I am confused.

JulioJerez, 2019-02-17:
ok I beleive the orginal bug dgBody::GetFirstContact is fixed now.
I was confused wondering whey I did not get it and it was because whe I call that function I do it form a callback. in teh call back the joint is always valid since is pass as the argument.

```
for (void* contact = NewtonContactJointGetFirstContact (contactJoint); contact; contact = NewtonContactJointGetNextContact (contactJoint, contact)) {
….
```

you are iteration over the contact joint on a body, ther is nothing wrong with that, because is part of teh interface,  is it only that this the style use by other libraries where the do teh update then quires teh scene for the result. I prefer the even driven system where you set a function to handle even wh eteh happens.  

what happen is that the engine is lazy, a body can have some deed joints that has no been deleted yet but that are inactive, this was is an optimization that try to eliminator the churning that happen when tow bodies come in close contact becaus ethe aabb overlap, but they are not colliding yet, in the case the engine crate the link between the two bodies by there is not joint. then of the body later collide teh joint is created,  but many time the body aabb are simple overlapping and no colliding.
adding a check for NULL fixit it.

Th eoethe problem that I saw is the assert on a contact normal that was not a unit vector, this will happen in debug and woul happen with non uniform scale, are you doing that.

to my knowledge the normal should be proper, by is possible that I missed one, in teh demo I see you have cones, and some oeth shape but I do not see the error by inspiriting the code.  
so when you sync, if the asserts happens in

`DG_INLINE dgVector dgCollisionInstance::SupportVertexSpecial (const dgVector& dir, dgInt32* const vertexIndex) const
`please let me know, the sould be reproducible and has nothing to do with the latest changes. In fact I cannot make it happen in teh sand box

JulioJerez, 2019-02-17:
I saw you close teh bug, did you get it running?

TrevorCash, 2019-02-17:
I saw you changed that code - I'll try again soon.  It was running before and was just very hard to replicate.

I'll reopen if I see any issues again because I think there are still a few issues I am having.  (I need to spend some time and get debug dlls working with the Urho3D cmake system again)

JulioJerez, 2019-02-17:
I though early you have set up to link to dlls,
That's usually good for sharing repro bugs.

TrevorCash, 2019-02-17:
Ok here is a crash I am getting, this is from the latest master branch:

In this particular demo I am calling NewtonBodySetMatrix() after every newton update.  Sometimes I get an error related to body equalibrium.  https://drive.google.com/open?id=1SW8vCpnxkuRLQ5lt151Df28Ua7pBIffw

TrevorCash, 2019-02-17:
I call NewtonBodySetSleepState before every call to NewtonBodySetMatrix()

JulioJerez, 2019-02-18:
I see a walking animation, it did not crash, do I have to do anything?

TrevorCash, 2019-02-18:
Oh right click i think to tilt the platform around

JulioJerez, 2019-02-18:
I see you call from outside
```
void dgBroadPhase::UpdateBody(dgBody* const body, dgInt32 threadIndex)
{
	if (m_rootNode && !m_rootNode->IsLeafNode() && body->m_masterNode) {
		dgBroadPhaseBodyNode* const node = body->GetBroadPhase();
		dgBody* const body1 = node->GetBody();
		dgAssert(body1 == body);
		dgAssert(!body1->m_equilibrium);

```

and that some how chaging the state of the body. 
Tell me this when you said crash, do you mean assert, of a hard crash that terminate the application.

TrevorCash, 2019-02-18:
An assert not a hard crash

TrevorCash, 2019-02-18:
The UpdateBody call originates from my call NewtonBodySetMatrix

TrevorCash, 2019-02-18:
NewtonBodySetMatrix
SetMatrixResetSleep
SetMatrix
UpdateCollisionMatrix
UpdateBody

JulioJerez, 2019-02-18:
Ah ok, thsi should be eassy to fix, basically you are moving a static body which should be in equilibrium by definition. 
basically the function
void NewtonBodySetMatrix(const NewtonBody* const bodyPtr, const dFloat* const matrixPtr)
should activate the body before calling SetMatrixResetSleep
I will fix it later.

TrevorCash, 2019-02-18:
I do have a call to
NewtonBodySetSleepState(newtonBody_, false);
right before NewtonBodySetMatrix. So maybe its not working all of the time?

 I added that early on in my integration to force wake the body.

JulioJerez, 2019-02-18:
Now I am confused, I added the call and and the assert still happens, which
means you must be running asyncrnous update, is this the case?

TrevorCash, 2019-02-18:
Ah! That was my mistake - it is asynchronous.  I just need to add a sync before the set matrix call!

JulioJerez, 2019-02-18:
yes you just call NewtonWaitForUpdateToFinish before you are going to change the state of a body.
this is why ther is the listener, that allowed you do that kind of game play logic at teh appropriate time.  

you do not really wnat to chnag ethe state of collision object at drawing time, that lead to jiter ansd strange behaviuor that seem liek bug, but is just state changes at teh wrong time.
check out some of the listerner demo ther are verety eassy to implement, and are not intrusive. tpo the engine. once you udnertan the Joints,  they are even eassier that that.

TrevorCash, 2019-02-18:
Got it, Makes sense - not sure why it did not occur to me earlier.

closed 2019-02-19 by TrevorCash