MADEAPPS/newton-dynamics #156: GCC compiling. Member with constructor not allowed in anonymous aggregate.

lewa-j, 2019-03-31:
Tested on GCC 8.1 on Windows x86_64 via MinGW-w64 and GCC 4.9 on Android Arm.

```
dgPhysics/dgWorldDynamicsParallelSolver.h:153:13: error: member 'dgVector dgWorkGroupFloat::<unnamed union>::<unnamed struct>::m_low' with constructor not allowed in anonymous aggregate
    dgVector m_low;
             ^~~~~
dgPhysics/dgWorldDynamicsParallelSolver.h:154:13: error: member 'dgVector dgWorkGroupFloat::<unnamed union>::<unnamed struct>::m_high' with constructor not allowed in anonymous aggregate
    dgVector m_high;
```

JulioJerez, 2019-04-04:
this is fixed now.
please sync and try again.

lewa-j, 2019-04-04:
Yes it helped, thanks. Can be closed.

closed 2019-04-05 by JulioJerez