MADEAPPS/newton-dynamics #158: Can't compile on Linux

schristy189, 2019-04-03:
Hi,

When running make I get the following error on dgMeshEffect1.cpp:

gcc version 6.3.0
cmake version 3.14.1
Debian Linux

```
[ 10%] Built target dgCore
[ 11%] Building CXX object sdk/dgPhysics/CMakeFiles/dgPhysics.dir/__/dgMeshUtil/dgMeshEffect1.cpp.o
In file included from /home/sam/code/newton/newton-dynamics/sdk/dgPhysics/./dgWorldDynamicUpdate.h:27:0,
                 from /home/sam/code/newton/newton-dynamics/sdk/dgPhysics/./dgWorld.h:32,
                 from /home/sam/code/newton/newton-dynamics/sdk/dgMeshUtil/dgMeshEffect1.cpp:24:
/home/sam/code/newton/newton-dynamics/sdk/dgPhysics/./dgWorldDynamicsParallelSolver.h:153:13: error: member ‘dgVector dgWorkGroupFloat::<anonymous union>::<anonymous struct>::m_low’ with constructor not allowed in anonymous aggregate
    dgVector m_low;
             ^~~~~
/home/sam/code/newton/newton-dynamics/sdk/dgPhysics/./dgWorldDynamicsParallelSolver.h:154:13: error: member ‘dgVector dgWorkGroupFloat::<anonymous union>::<anonymous struct>::m_high’ with constructor not allowed in anonymous aggregate
    dgVector m_high;
             ^~~~~~
sdk/dgPhysics/CMakeFiles/dgPhysics.dir/build.make:62: recipe for target 'sdk/dgPhysics/CMakeFiles/dgPhysics.dir/__/dgMeshUtil/dgMeshEffect1.cpp.o' failed
make[2]: *** [sdk/dgPhysics/CMakeFiles/dgPhysics.dir/__/dgMeshUtil/dgMeshEffect1.cpp.o] Error 1
CMakeFiles/Makefile2:284: recipe for target 'sdk/dgPhysics/CMakeFiles/dgPhysics.dir/all' failed
make[1]: *** [sdk/dgPhysics/CMakeFiles/dgPhysics.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

JulioJerez, 2019-04-04:
oh I fixed this one  <anonymous struct>::m_low’ with constructor not allowed in anonymous aggregate
but removing the union, and using casting in the code.

the secund one,  recipe for target 'sdk/dgPhysics/CMakeFiles/dgPhysics.dir....
I do not know what that is maybe is triggered by the first one.
please try again.

schristy189, 2019-04-04:
Thanks, that fixed the first issue, I am now getting this error:

```
[ 11%] Built target dgCore
[ 31%] Built target dgPhysics
[ 32%] Built target newton
[ 34%] Built target dMath
[ 38%] Built target dContainers
[ 49%] Built target dCustomJoints
[ 49%] Building CXX object sdk/dNewton/CMakeFiles/dNewton.dir/dNewton.cpp.o
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp: In member function ‘void dNewton::ResetTimer()’:
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:241:17: error: ‘CLOCK_REALTIME’ was not declared in this scope
   clock_gettime(CLOCK_REALTIME, &ts); // Works on Linux
                 ^~~~~~~~~~~~~~
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:241:36: error: ‘clock_gettime’ was not declared in this scope
   clock_gettime(CLOCK_REALTIME, &ts); // Works on Linux
                                    ^
In file included from /home/sam/code/newton/newton-dynamics/sdk/dNewton/dStdAfxNewton.h:61:0,
                 from /home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:22:
/home/sam/code/newton/newton-dynamics/sdk/dgNewton/./Newton.h:50:16: error: expected primary-expression before ‘long’
  #define dLong long long
                ^
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:243:17: note: in expansion of macro ‘dLong’
   m_baseCount = dLong (ts.tv_sec) * 1000000 + ts.tv_nsec / 1000;
                 ^~~~~
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp: In member function ‘long long int dNewton::GetTimeInMicrosenconds() cons
’:
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:272:17: error: ‘CLOCK_REALTIME’ was not declared in this scope
   clock_gettime(CLOCK_REALTIME, &ts); // Works on Linux
                 ^~~~~~~~~~~~~~
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:272:36: error: ‘clock_gettime’ was not declared in this scope
   clock_gettime(CLOCK_REALTIME, &ts); // Works on Linux
                                    ^
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:275:31: error: ‘unsigned64’ was not declared in this scope
   return unsigned64 (ts.tv_sec) * 1000000 + ts.tv_nsec / 1000 - m_baseCount;
                               ^
sdk/dNewton/CMakeFiles/dNewton.dir/build.make:75: recipe for target 'sdk/dNewton/CMakeFiles/dNewton.dir/dNewton.cpp.o' failed
make[2]: *** [sdk/dNewton/CMakeFiles/dNewton.dir/dNewton.cpp.o] Error 1
CMakeFiles/Makefile2:399: recipe for target 'sdk/dNewton/CMakeFiles/dNewton.dir/all' failed
make[1]: *** [sdk/dNewton/CMakeFiles/dNewton.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

JulioJerez, 2019-04-04:
I made it resturn 0 for now, until we get how to read teh timer on linux.

//#if (defined (_POSIX_VER) || defined (_POSIX_VER_64))
#ifdef __linux__
        //timespec ts;
//clock_gettime(CLOCK_REALTIME, &ts); // Works on Linux
//return unsigned64 (ts.tv_sec) * 1000000 + ts.tv_nsec / 1000 - m_baseCount;
return 0;
#endif

please try again.

schristy189, 2019-04-04:
Thanks, it looks like just just have to include time.h in dNewton.cpp and it should work.

Now I get the following error:

```
[ 11%] Built target dgCore
[ 31%] Built target dgPhysics
[ 32%] Built target newton
[ 34%] Built target dMath
[ 38%] Built target dContainers
[ 49%] Built target dCustomJoints
[ 56%] Built target dNewton
[ 58%] Built target tinyxml
[ 70%] Built target dScene
[ 72%] Built target dAnimation
[ 76%] Built target dVehicle
[ 78%] Built target imgui
[ 78%] Built target glatter
[ 78%] Building CXX object sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o
In file included from /home/sam/code/newton/newton-dynamics/sdk/dgNewtonAvx/dgSolver.cpp:23:0:
/home/sam/code/newton/newton-dynamics/sdk/dgNewtonAvx/dgSolver.h: In constructor ‘dgSoaFloat::dgSoaFloat(const dgVector&, const dgVector&)’:
/home/sam/code/newton/newton-dynamics/sdk/dgNewtonAvx/dgSolver.h:197:50: error: ‘_mm256_loadu2_m128’ was not declared in this scope
   : m_type(_mm256_loadu2_m128(&high.m_x, &low.m_x))
                                                  ^
sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/build.make:75: recipe for target 'sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o'
failed
make[2]: *** [sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o] Error 1
CMakeFiles/Makefile2:938: recipe for target 'sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/all' failed
make[1]: *** [sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

JulioJerez, 2019-04-04:
ah it seem that intrinsic seem to be a macro the resolve to this
	DG_INLINE dgSoaFloat(const dgVector& low, const dgVector& high)
//		:m_type(_mm256_loadu2_m128(&high.m_x, &low.m_x))
		:m_type(_mm256_set_m128 (high.m_type, low.m_type))

It is fixed now, please try again.

JulioJerez, 2019-04-06:
did that last change fixed the problem?

schristy189, 2019-04-09:
Sorry for the delay! This is the output I get now:

```
[ 11%] Built target dgCore
[ 31%] Built target dgPhysics
[ 32%] Built target newton
[ 34%] Built target dMath
[ 38%] Built target dContainers
[ 49%] Built target dCustomJoints
[ 49%] Building CXX object sdk/dNewton/CMakeFiles/dNewton.dir/dNewton.cpp.o
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp: In member function ‘void dNewton::ResetTimer()’:
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:243:3: error: ‘baseCount’ was not declared in this scope
   baseCount = ts.tv_nsec / 1000;
   ^~~~~~~~~
In file included from /home/sam/code/newton/newton-dynamics/sdk/dNewton/dStdAfxNewton.h:63:0,
                 from /home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:22:
/home/sam/code/newton/newton-dynamics/sdk/dgNewton/./Newton.h:50:16: error: expected primary-expression before ‘long’
  #define dLong long long
                ^
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:244:17: note: in expansion of macro ‘dLong’
   m_baseCount = dLong (ts.tv_sec) * 1000000 + ts.tv_nsec / 1000;
                 ^~~~~
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp: In member function ‘long long int dNewton::GetTimeInMicrosenconds() cons
’:
/home/sam/code/newton/newton-dynamics/sdk/dNewton/dNewton.cpp:271:31: error: ‘unsigned64’ was not declared in this scope
   return unsigned64 (ts.tv_sec) * 1000000 + ts.tv_nsec / 1000 - m_baseCount;
                               ^
sdk/dNewton/CMakeFiles/dNewton.dir/build.make:75: recipe for target 'sdk/dNewton/CMakeFiles/dNewton.dir/dNewton.cpp.o' failed
make[2]: *** [sdk/dNewton/CMakeFiles/dNewton.dir/dNewton.cpp.o] Error 1
CMakeFiles/Makefile2:399: recipe for target 'sdk/dNewton/CMakeFiles/dNewton.dir/all' failed
make[1]: *** [sdk/dNewton/CMakeFiles/dNewton.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

JulioJerez, 2019-04-09:
ok I made return zero again until we figure out what is teh name of teh header that include timespec for all Linux systems. In Unbutu time.h seem to do it. 
anyway can you try Again?

schristy189, 2019-04-12:
Now I get this error:
```
[ 11%] Built target dgCore
[ 31%] Built target dgPhysics
[ 32%] Built target newton
[ 34%] Built target dMath
[ 38%] Built target dContainers
[ 49%] Built target dCustomJoints
[ 56%] Built target dNewton
[ 58%] Built target tinyxml
[ 70%] Built target dScene
[ 72%] Built target dAnimation
[ 76%] Built target dVehicle
[ 78%] Built target imgui
[ 78%] Built target glatter
[ 78%] Building CXX object sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o
In file included from /home/sam/code/newton/newton-dynamics/sdk/dgNewtonAvx/dgSolver.cpp:23:0:
/home/sam/code/newton/newton-dynamics/sdk/dgNewtonAvx/dgSolver.h: In constructor ‘dgSoaFloat::dgSoaFloat(const dgVector&, const dgVector&)’:
/home/sam/code/newton/newton-dynamics/sdk/dgNewtonAvx/dgSolver.h:200:51: error: ‘_mm256_set_m128’ was not declared in this scope
   :m_type(_mm256_set_m128 (high.m_type, low.m_type))
                                                   ^
sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/build.make:75: recipe for target 'sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o'
failed
make[2]: *** [sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o] Error 1
CMakeFiles/Makefile2:938: recipe for target 'sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/all' failed
make[1]: *** [sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
Also, time.h is standard for all Linux systems. See the C99 standard (ISO/IEC 9899) section 7.23. You can find drafts for the C and C++ standards [here.](https://stackoverflow.com/a/83763)

JayFoxRox, 2019-04-12:
I can't reproduce this issue locally using GCC 8.2.1.
A simple `mkdir build; cd build; cmake ..; make` does the job.

---

`_mm256_set_m128` was added 2 years ago in GCC: https://github.com/gcc-mirror/gcc/commit/23f05e90ea5b60b676c69f5bf481bfd6c3a90160

GCC 6.x is from 2016, so this intrinsic didn't exist back then.
I'm not sure if it's worth supporting such an old compiler if free alternatives exist.

Anyhow, a workaround appears to be explained in OpenCV:

https://github.com/opencv/opencv/blob/5cb0eded642861d0f845f9c32140b00afd15826a/modules/calib3d/src/undistort.avx2.cpp#L174

---

The easiest workaround for your issue is to simply disable the AVX Plugin like this: `cmake .. -DNEWTON_WITH_AVX_PLUGIN=OFF`

I'd assume you might want to enable `NEWTON_WITH_SSE4_PLUGIN` as an alternative for speedups, when not using AVX.

JulioJerez, 2019-04-12:
so gcc does not support _mm256_set_m128 either? I see if I can find an alternative way this weekend.

JayFoxRox, 2019-04-12:
> so gcc does not support _mm256_set_m128 either?

See the tags in the commit I linked: (at least) gcc-8_3_0-release gcc-8_2_0-release gcc-8_1_0-release support it. It's possible that gcc 7.x or earlier also supports it.

It definitely works fine on my GCC 8.2.1.

>  I see if I can find an alternative way this weekend.

@schristy189 is using 2 year old software. I don't feel like this must be supported in an optional part of newton-dynamics. They can upgrade to GCC 8.x for free or use an alternative toolchain.

If you really want to implement an alternative you should probably add a check for `#if __GNUC__ <= 7` (or 6?) for this workaround, to not sacrifice performance on up-to-date compilers. An alternative workaround has also been linked from my previous post (analog to what OpenCV does, using `_mm256_setzero_ps` and 2x `_mm256_insertf128_ps`, which should be supported since GCC 5.x, as implied by the OpenCV comment).

JulioJerez, 2019-04-12:
well it turned out that function was only used in one place. It was very eassy to replace it with SSE equivalent. if you sync, it should build now.

schristy189, 2019-04-14:
Thanks! I have no problem using an alternative GCC, I only used this version because it is the default Debian version. For the stable release of Debian, the software can often be 1-3 years behind the "bleeding edge" software versions, for instance the default cmake for Debian is version 3.7.1 while the current release is 3.14.1.

I'll be compiling this project with gcc-8 and g++-8 from now on for compatibility.

Here is the error I am getting now:

```
[ 34%] Built target dMath
[ 38%] Built target dContainers
[ 49%] Built target dCustomJoints
[ 56%] Built target dNewton
[ 58%] Built target tinyxml
[ 70%] Built target dScene
[ 72%] Built target dAnimation
[ 76%] Built target dVehicle
[ 78%] Built target imgui
[ 78%] Built target glatter
[ 79%] Built target dgNewtonAvx
[ 79%] Building CXX object applications/demosSandbox/CMakeFiles/demosSandbox.dir/sdkDemos/demos/DynamicRagdoll.cpp.o
In file included from /home/sam/code/newton/newton-dynamics/applications/demosSandbox/../../sdk/dMath/dVector.h:16,
                 from /home/sam/code/newton/newton-dynamics/applications/demosSandbox/sdkDemos/toolBox/toolbox_stdafx.h:120,
                 from /home/sam/code/newton/newton-dynamics/applications/demosSandbox/sdkDemos/demos/DynamicRagdoll.cpp:12:
/home/sam/code/newton/newton-dynamics/applications/demosSandbox/sdkDemos/demos/DynamicRagdoll.cpp: In member function ‘virtual void dAnimationHipEffector::SubmitConstraints(float, int)’:
/home/sam/code/newton/newton-dynamics/applications/demosSandbox/sdkDemos/demos/DynamicRagdoll.cpp:148:28: error: ‘pitchMatrix’ was not declared in this scope
     dFloat angle = dAtan2 (pitchMatrix[1][2], pitchMatrix[1][1]);
                            ^~~~~~~~~~~
/home/sam/code/newton/newton-dynamics/applications/demosSandbox/../../sdk/dMath/dMathDefines.h:91:43: note: in definition of macro ‘dAtan2’
 #define dAtan2(x,y) dFloat (atan2 (dFloat(x), dFloat(y)))
                                           ^
/home/sam/code/newton/newton-dynamics/applications/demosSandbox/sdkDemos/demos/DynamicRagdoll.cpp:148:28: note: suggested alternative:
dPitchMatrix’
     dFloat angle = dAtan2 (pitchMatrix[1][2], pitchMatrix[1][1]);
                            ^~~~~~~~~~~
/home/sam/code/newton/newton-dynamics/applications/demosSandbox/../../sdk/dMath/dMathDefines.h:91:43: note: in definition of macro ‘dAtan2’
 #define dAtan2(x,y) dFloat (atan2 (dFloat(x), dFloat(y)))
                                           ^
applications/demosSandbox/CMakeFiles/demosSandbox.dir/build.make:361: recipe for target 'applications/demosSandbox/CMakeFiles/demosSandbox.dir/sdkDemos/demos/DynamicRagdoll.cpp.o' failed
make[2]: *** [applications/demosSandbox/CMakeFiles/demosSandbox.dir/sdkDemos/demos/DynamicRagdoll.cpp.o] Error 1
CMakeFiles/Makefile2:1005: recipe for target 'applications/demosSandbox/CMakeFiles/demosSandbox.dir/all' failed
make[1]: *** [applications/demosSandbox/CMakeFiles/demosSandbox.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

If anybody comes across this post and doesn't know how to upgrade gcc, here are the instructions:

```
sudo echo deb http://ftp.us.debian.org/debian sid main >> /etc/apt/sources.list
sudo apt update; sudo apt install gcc-8 g++-8
export CXX=g++-8
export CC=gcc-8
```

Then just make the project like you normally would.

JulioJerez, 2019-04-14:
it should compile now, I removed that intrinsic altogether.  but is good you upgrade to newer compiler
as for the atan2 error, I believe it is  was fixed now.

schristy189, 2019-04-15:
I now get the following error, removing the applications/media/ directory allows me to complete compilation:

```
[ 10%] Built target dgCore
[ 29%] Built target dgPhysics
[ 30%] Built target newton
[ 33%] Built target dMath
[ 36%] Built target dContainers
[ 47%] Built target dCustomJoints
[ 54%] Built target dNewton
[ 56%] Built target tinyxml
[ 67%] Built target dScene
[ 70%] Built target dAnimation
[ 74%] Built target dVehicle
[ 75%] Built target imgui
[ 76%] Built target glatter
[ 78%] Built target dgNewtonAvx
[ 79%] Linking CXX executable demosSandbox
symbolic link resources folder from /home/sam/code/newton/newton-dynamics/applications/media => /home/sam/code/newton/newton-dynamics/applications/media
failed to create symbolic link '/home/sam/code/newton/newton-dynamics/applications/media' because existing path cannot be removed: Is a
directory
applications/demosSandbox/CMakeFiles/demosSandbox.dir/build.make:1026: recipe for target 'applications/demosSandbox/demosSandbox' failed
make[2]: *** [applications/demosSandbox/demosSandbox] Error 1
make[2]: *** Deleting file 'applications/demosSandbox/demosSandbox'
CMakeFiles/Makefile2:1005: recipe for target 'applications/demosSandbox/CMakeFiles/demosSandbox.dir/all' failed
make[1]: *** [applications/demosSandbox/CMakeFiles/demosSandbox.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

JulioJerez, 2019-04-16:
I do not why are all these error,  when I tryied thsi in Ubuntu Linux is worked.
It seem difrent ditribution work diffrently. 
I can; no test my unbutu laptop because wit the lates ubuntu I keep getting thsi time out, and I get conflicting explantion as how to fix that problem

but ion any case, in window we nee to run the Install option, is no automatic

I think you have to run the make install. the sandbox is probably trying to get the shaders from the media folder.

JayFoxRox, 2019-04-16:
> I do not why are all these error, when I tryied thsi in Ubuntu Linux is worked.
> It seem difrent ditribution work diffrently.

No, that's not the issue here. This code tries to symlink the media folder, but it doesn't check if that folder already exists:

https://github.com/MADEAPPS/newton-dynamics/blob/bed7292e3374ab9c3c9b942a6bd4e9e03de5cd42/applications/demosSandbox/CMakeLists.txt#L71-L81

So @schristy189 had a failed build, which had already done this step, but then stopped because of other bugs. When those other bugs were fixed, they just tried to `make` again without removing the garbage left behind by the previous build. So CMake fails to create the same link again, because it already exists.

They worked around this issue by removing that target folder (meaning CMake is able to symlink it again, without errors):

> removing the applications/media/ directory allows me to complete compilation

It would have worked fine if they had created a new build-directory / restarted from scratch.
But this is still an issue in the newton build-system: it shouldn't be affected by previous build-failures.

This issue can be closed; a new one for the build-system issue should be opened (I won't do it).

closed 2019-05-04 by JulioJerez

StuAtGit, 2022-09-27:
Did anyone ever open a bug for a failed build wedging any further builds, due to the existing media directory issue? Just asking before I do it.. I did search, but didn't find one.