MADEAPPS/newton-dynamics #159: Cannot compile on Linux when using double precision

Solenius, 2019-04-08:
I have the following issue when trying build on Linux (Ubuntu) platform.
GCC 8.2.0
cmake version 3.12.1

```
In file included from /sdk/dgNewtonAvx/dgSolver.cpp:23:
/sdk/dgNewtonAvx/dgSolver.h: In constructor ‘dgSoaFloat::dgSoaFloat(const dgBigVector&, const dgBigVector&)’:
/sdk/dgNewtonAvx/dgSolver.h:62:10: error: ‘_mm256_loadu2_m128d’ was not declared in this scope
   :m_low(_mm256_loadu2_m128d(&low.m_z, &low.m_x))
          ^~~~~~~~~~~~~~~~~~~
/sdk/dgNewtonAvx/dgSolver.h:62:10: note: suggested alternative: ‘_mm256_loadu_si256’
   :m_low(_mm256_loadu2_m128d(&low.m_z, &low.m_x))
          ^~~~~~~~~~~~~~~~~~~
          _mm256_loadu_si256
make[2]: *** [sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/build.make:76: sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/dgSolver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:884: sdk/dgNewtonAvx/CMakeFiles/dgNewtonAvx.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 13%] Building CXX object sdk/dgNewtonSse4.2/CMakeFiles/dgNewtonSse4.2.dir/dgWorldBase.cpp.o
[ 13%] Building CXX object sdk/dgNewtonSse4.2/CMakeFiles/dgNewtonSse4.2.dir/dllmain.cpp.o
[ 14%] Building CXX object sdk/dgCore/CMakeFiles/dgCore.dir/dgConvexHull4d.cpp.o
In file included from /sdk/dgNewtonAvx2/dgSolver.cpp:23:
/sdk/dgNewtonAvx2/dgSolver.h: In constructor ‘dgSoaFloat::dgSoaFloat(const dgBigVector&, const dgBigVector&)’:
/sdk/dgNewtonAvx2/dgSolver.h:63:10: error: ‘_mm256_loadu2_m128d’ was not declared in this scope
   :m_low(_mm256_loadu2_m128d(&low.m_z, &low.m_x))
          ^~~~~~~~~~~~~~~~~~~
/sdk/dgNewtonAvx2/dgSolver.h:63:10: note: suggested alternative: ‘_mm256_loadu_si256’
   :m_low(_mm256_loadu2_m128d(&low.m_z, &low.m_x))
          ^~~~~~~~~~~~~~~~~~~
          _mm256_loadu_si256
make[2]: *** [sdk/dgNewtonAvx2/CMakeFiles/dgNewtonAvx2.dir/build.make:76: sdk/dgNewtonAvx2/CMakeFiles/dgNewtonAvx2.dir/dgSolver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:939: sdk/dgNewtonAvx2/CMakeFiles/dgNewtonAvx2.dir/all] Error 2
```

JulioJerez, 2019-04-08:
fixed. 
so you are testing parallel solve in double with linux, awesome!!

Solenius, 2019-04-08:
Works like a charm, thank you :+1:

closed 2019-04-08 by Solenius