MADEAPPS/newton-dynamics #164: Recreate Bullet's setAngularFactor(0.0f)

dustContributor, 2019-04-14:
Hi! I'm trying to find out how it'd be possible to do the following in Newton:

In Bullet I had the "player" character attached to a "standing" capsule rigid body, that I could move around by manipulating its velocity.

This worked because I could do body->setAngularFactor(0.0f), which prevented Bullet from rotating the body in the simulation. Bullet could still simulate collision, so the player entity would fall and collide normally, but it maintain the capsule in the original "standing" position.

Is it possible to do the same in Newton?

JulioJerez, 2019-04-14:
I think you are taking about kinematic objects.
In newton you can create kinematic objects that has some of the property of dynamics and some o fthe property of static.
these objects, the application is responsible for moving them. the application set the velocity an dthe call integrate to moves them around.
for thsi ther are this classes call listener that are like mini system that allows the end user to implement this own logic as to how they want to control a body.

take a look are the demo Kinematic objects that manipulate two moving platforms, that interactt with objects in the scene. 
so the answer is yes, you can object that you can move around by setting velocity and integrate, and never set the angula velocity.

but the question is if you have your system set with othe engine why are you try to do the same with newton? Newton has a learning curve that, judging by your question, setting up why you want required some more work than just calling a junction on an object like setangularfactor, what evet that is and expect that you work. 
Things like that then not to work in newton unless they are set the proper way so that the engine do no see conflicts. liek ex using a kinematic body

dustContributor, 2019-04-14:
> I think you are taking about kinematic objects.

Sort of. What I got from the kinematic object docs is that they don't collide. That is, you get notified if they collide, but the physics simulation wont prevent them from going through a wall so to speak. I'd end up doing the physics simulation for the kinematic objects myself, which is not something I'd want to do. I don't know physics, Newton does, I'd like to leave it to Newton.

Bullet allowed me to prevent it from rotating the object, while at the same time allowing me to manipulate its velocity to move it around, and with Bullet simulating gravity/collision with it without issues.

Would I be able to do that with kinematic objects?

The other idea I haven't tried is to reset the rotation part of the object's transform each simulation step and see how it works out.

I don't need it to be perfect, I just want to avoid having to implement my own crappy physics simulation just for player movement that I know I wont get right until I re-implement it a dozen times.

> but the question is if you have your system set with othe engine why are you try to do the same with newton?

That's as far as my physics engine integration went so it isn't that much work to move from Bullet to Newton. Just plain rigid bodies (boxes, spheres, capsules) and a pseudo kinematic body for the camera movement.

It's a long story but basically I'm using Newton from Java, and I have the choice between using crappy Bullet C++ API bindings that have a bunch of dependencies that I don't want, or good Newton C API bindings. 

I'd rather use the better Newton bindings than pulling all of this other crap I don't want just to use Bullet.

JulioJerez, 2019-04-14:
"What I got from the kinematic object docs is that they don't collide. "
that's one way that the work. they other is that the are just like dynamics bodies that collide with other dynamics bodies, by that they do no get integrated. but like I said the is not calling some function one  body to change the behavior for a body in newton. if you can play the sandbox demos, and you select from the menu Kinematic bodies  you will see two moving platform both made with kinematic bodies, one collide and the other does not, it simple generate a contact joint.. 
you still have to do some work to get it going, for that the engine have listener, which are way to plug and arbitrary callback that the engine will call and you can apply you game logic.
here in the pre update you set the linear velocity  and do no set angular velocity.
the in the post update you call integrate body to move the object around.
below is the sample code, you can just check the demo, is aby 200 lines of code. 

void PreUpdate(dFloat timestep) 
	{
....
			NewtonBodySetVelocity(entry.m_plaform, &veloc[0]);

			UpdateBodies(entry);
		}
	}

	virtual void PostUpdate(dFloat timestep) 
	{
		for (dList<KinematicPlatform>::dListNode* ptr = m_platformList.GetFirst(); ptr; ptr = ptr->GetNext()) {
			const KinematicPlatform& entry = ptr->GetInfo();
			NewtonBodyIntegrateVelocity (entry.m_plaform, timestep);
		}
	}

as for setting a flag body of over riding the matrix, I am afraid that  answer is not that's not how the newton engine word. 

before you embark of a task that may take over many unknown, I suggest to download the SDK an run the demos, see if that satisfy your needs,  

you are correct, making players is difficult, There is a play controller in the game that is probably broken and required to re enabled again, I can try re enable if you are interested.  this will be like sting it up and using it more less they we you want,

dustContributor, 2019-04-22:
I see. I managed to compile and execute the demos, and that "solid" kinematic body seems to be what I want.

So far from what I've seen in the code the steps are:
1. Create collider (sphere in this case)
2. Create kinematic body with the collider
3. Attach force/torque callback (I think? It's to apply gravity)
4. Set collision mode to 1 for the collider

Then each update/frame/tick:
1. Integrate velocity from previous update.
2. Copy back transformation changes from Newton (rotation/translation).
3. Set new velocity.

Am I missing something? As-is, when I set a new velocity then integrate it, the position from the transform returned by NewtonBodyGetMatrix doesn't seems to change, so I end up with an immobile body.

JulioJerez, 2019-04-22:
make a listener an hold to a list or vector that you want to control that way.
Then each update/frame/tick 
1.  in the pre listener callback  for each body on that list set velocity
2.  in the post listener callback for each body call Inegrate velocity

basically in the prelisner you set the velocity so that the colslion system know how to interact with oth bodies. them in the post listener, integrate velocity is what moved the object an update the collision system.
all that happen within a newtion update. the you just read the transform

dustContributor, 2019-04-24:
You mean with NewtonSetPostUpdateCallback function?

I don't see a NewtonSetPreUpdateCallback function. I do see a NewtonWorldListenerSetPreUpdateCallback though. Which ones should I be using?

JulioJerez, 2019-04-24:
Do you use cpp?
If you set the custom joint library, there is a base class that you
subclass from,
 it does all the house keeping work to set the callbacks.

Them you just implement yourpre and post update methods on your subclass.

It is how the kinematic demo and all other demos do it.

dustContributor, 2019-04-24:
> Do you use cpp?

No, like I said before, I use the C API from Java.

Which callback functions should I be using?

JulioJerez, 2019-04-24:
Oh I see. In that case you can take the class
dCustomListener::dCustomListener(NewtonWorld* const world, const char*
const listenerName)
:dCustomAlloc()
,m_world(world)
{
void* const listener = NewtonWorldAddListener (world, listenerName, this);
NewtonWorldListenerSetDestructorCallback (world, listener, Destroy);
NewtonWorldListenerSetPreUpdateCallback (world, listener, PreUpdate);
NewtonWorldListenerSetPostUpdateCallback (world, listener, PostUpdate);
NewtonWorldListenerSetBodyDestroyCallback (world, listener, OnDestroyBody);
}

and see how the callbacks are set, ther are four and there are all
optional. I just implement the all.
you can make your own Java class and the use that as teh vase fore your
listeners.

Listeners are useful to implements sub system that group objects and apply
the same operation to all of them

JulioJerez, 2019-04-25:
If you are interested, the is an automatic wrapper generator base on swing,
that we used for making the chap plugin, is is eassy to wrap the classes
you need an make a have wrapper.

If you are interested I can set the I,igual steps, and you can continue
from there on you own.
This makes a wrapper that to you will look as if the engine was written in
java.

dustContributor, 2019-04-29:
Thanks for the offer but I am using LWJGL to make bindings for the C API. I actually prefer the direct access to the C API.

I think I understand better the problem, there are two ways to implement character controllers:

1. Kinematic, in which the physics engine generates contact/collision information, but the user is responsible for simulating the body's response. This is what you're doing in the kinematic body example, except for the fact that the platforms don't respond to collisions, you have one platform that is fully transparent and another that can push other bodies around, but the platform itself is unaffected by the physics engine.
 
2. Dynamic, in which by tweaking the body's properties you can let the physics engine resolve collision responses, while at the same time letting the user set the body's velocity to move it around.

For the second approach to be possible, the engine should allow the user to disable certain aspects of the simulation, like what you can do in Bullet by setting the angular factor to zero, in such way that the simulation wont rotate the body, but it will simulate collision with other bodies, gravity and so on.

I've looked around and I think it isn't possible to do the second approach in Newton. I can make a dynamic rigid body, but it exhibits weird behavior when setting its velocity, and it's impossible for me to tell the engine not to rotate the body unless doing something drastic like `NewtonBodySetMassMatrix(body, mass, VERY_HIGH_NUMBER, VERY_HIGH_NUMBER, VERY_HIGH_NUMBER)`

Even then it doesn't works that well, if I try to apply a constant gravity force to it, the body will twitch trying to go through the ground and getting moved back again.

So what I'm left with is to make a kinematic controller, simulating the sliding collision response myself.

Am I correct?

JulioJerez, 2019-04-29:
tricks like NewtonBodySetMassMatrix(body, mass, VERY_HIGH_NUMBER, VERY_HIGH_NUMBER, VERY_HIGH_NUMBER) do no work in newton.

In newton if you wna to make a dynamic player, you just make a dynamics body liek any other and you apply a 2d joint, the you are free to so all the manipulation you wnat to the joint. 
In the engine the joint is dCustomUpVector, It will constraint the body to be upright and you can 

you can also use the dCustomKinematicController which is probably what you want, this wan will simple place the player what you want by setting the matrix. just like that
it will interact with world naturally without explosion or collision failing. 

The question is what type of controller you want to make?

dustContributor, 2019-05-01:
So for a dynamic body you're suggesting something like this:

[image]

Say that I want to model it as an upright capsule or cylinder with a sphere as a base, so I set the sphere as the parent, the capsule as the child, and that join there that would keep the capsule or cylinder upright.

Honestly I want the easiest to implement probably. So far I'm not convinced which one will be easier, this constraint stuff seems complex.

JulioJerez, 2019-05-01:
“Honestly I want the easiest to implement probably. So far I'm not
convinced which one will be easier, this constraint stuff seems complex.”

That’s because you are used to how other engine work.  I did no say it was
easy, I said this how you use dynamics bodies in newton to get then to do
what you want.

I am not sure what it is that you want do.

There is a character controlled in the library that is based on a kinematic
bodies. But is in the dCustomJointlibrary, which is  c++, you said you do
not want to deal with cPP,

The player controller is probably no working ow but si a matter to
uncomment out and fix the syntax errors.

Has you wrapped any of the joints from the joint library.

JulioJerez, 2019-05-01:
the Kinematic constrain joint, is attache to the body only, the to move the
body you just set the target matrix and it will move there if it can.

dustContributor, 2019-05-02:
>  I am not sure what it is that you want do.

You're right, I'm tripping myself over. Let's start again. I want to make a kinematic body that pushes objects around.

Following your code here:
https://github.com/MADEAPPS/newton-dynamics/blob/master/applications/demosSandbox/sdkDemos/demos/KinematicBodies.cpp

I'm going to simplify a bit the code since I have a lot of fluff that isn't relevant, here is what I do:

```java
var collider = NewtonCreateBox(world, p0, p1, p2, 0, identity);
var body = NewtonCreateKinematicBody(world, collider, wmat);
NewtonDestroyCollision(collider);
collider = NewtonBodyGetCollision(body);
// If I dont set the mass, I can't move the kinematic body with IntegrateVelocity
NewtonBodySetMassProperties(body, mass, collider);
NewtonCollisionSetMode(collider, 1);
```

There I have the kinematic body with collision mode set to 1, as per your example. Then I have the two callbacks you mentioned:
```java
NewtonWorldListenerSetPreUpdateCallback( world, newtonListener, (w,d, timestep ) -> {
	for ( int i = 0; i < entities.size(); ++i ) {
		var eid = entities.get(i);
		var b = bodies.get( eid );
		var vel = velocities.get( eid );
		NewtonBodySetVelocity( b.body, buffer
				.put( 0, vel.next.x )
				.put( 1, vel.next.y )
				.put( 2, vel.next.z ) );
	}
} );
NewtonWorldListenerSetPreUpdateCallback( world, newtonListener, (w,d, timestep ) -> {
	for ( int i = 0; i < entities.size(); ++i ) {
		var eid = entities.get(i);
		var b = bodies.get( eid );
		NewtonBodyIntegrateVelocity( b.body, timestep );
	}
} );
```

Here is what happens: https://gfycat.com/InformalEdibleFlyingfox

The floating box is the kinematic one, the falling box is a normal dynamic body. From what I saw in your kinematic demo, the falling box should collide with the kinematic one, do you know what could I be doing wrong?

JulioJerez, 2019-05-02:
oh I, over the weekend I can enable the player controller manager.  there
is another use that just ask for a player controller.
I based on these class

*class dCustomPlayerControllerManager;*
*class dCustomPlayerController*
*{*
* public:*
* void* m_userData;*
* NewtonBody* m_kinematicBody;*
* dCustomPlayerControllerManager* m_manager;*
*};*

*class dCustomPlayerControllerManager: public dCustomParallelListener*
*{*
* public:*
* CUSTOM_JOINTS_API dCustomPlayerControllerManager(NewtonWorld* const
world);*
* CUSTOM_JOINTS_API ~dCustomPlayerControllerManager();*

*// CUSTOM_JOINTS_API virtual void ApplyPlayerMove(dCustomPlayerController*
const controller, dFloat timestep) = 0;*
* CUSTOM_JOINTS_API virtual dCustomPlayerController* CreatePlayer(dFloat
mass, dFloat outerRadius, dFloat innerRadius, dFloat height, dFloat
stairStep, const dMatrix& localAxis);*
* CUSTOM_JOINTS_API virtual int ProcessContacts(const
dCustomPlayerController* const controller, NewtonWorldConvexCastReturnInfo*
const contacts, int count) const;*

* protected:*
* virtual void PreUpdate(dFloat timestep) {}*
* CUSTOM_JOINTS_API virtual void PostUpdate(dFloat timestep, int threadID);*

* dList<dCustomPlayerController> m_playerList;*
* dFloat m_timestep;*
*};*

and will be the demo
../../applications\demosSandbox\sdkDemos\demos\BasicPlayerController.cpp

but you have to be able to intefrate CPP so that you can use the
dCustomJoint Library,  using newton in C is too low level.

dustContributor, 2019-05-02:
Cool! I'll check it out later when I get to that point. I can probably implement it in Java using the C API if I understand the code.

Could you please answer my previous question?

JulioJerez, 2019-05-03:
It does no even need the listener this part 

```
var collider = NewtonCreateBox(world, p0, p1, p2, 0, identity);
var body = NewtonCreateKinematicBody(world, collider, wmat);
NewtonDestroyCollision(collider);
collider = NewtonBodyGetCollision(body);
// If I dont set the mass, I can't move the kinematic body with IntegrateVelocity
NewtonBodySetMassProperties(body, mass, collider);
NewtonCollisionSetMode(collider, 1);

```
should make a collidable body,  I will try to past that in the sand box and see what happens.

the last past I assume you meant PostUpdate no preupdate
NewtonWorldListenerSetPreUpdateCallback( world, newtonListener, (w,d, timestep ) -> {

In the mean time, I start the revamping of the player controller, I set teh classes and everything, teh player just fall by gravity, I have no done the collsion loop yet.

I suggest you sync and try out so that you can see the process hwo to work wit the engine.
Play player controller is no so trivial, and in your case you will be dealing with three issues:
the NewtonLibrary,  Implementing a new functionalisty and teh porting issue to java.

I have neglected the player controller for a long time so is time to revisit and make functional again.
so if you check it out and see how it work and make progress you have a less difficult time.

once you get the basic of Newton them it become simpler to make progress.
I will continue the collision part later.

JulioJerez, 2019-05-03:
Ok I now added enough that you can see how the final player will behave.
The par missing is the contact resolution solver, and the from the is all
on the use side to control it however he wants.

dustContributor, 2019-05-04:
It was an old build, **that** was the issue :sob: :sob: :sob: 

I had these pre-compiled binaries from 2-3 months ago. I updated to a new build, now it works like in the demos. I am really angry that I wasted so much time in it, but I am really happy that it works now. 

Anyway, now I can check out your player controller and some sliding behavior papers around to implement actual movement.

Thank you for your time.

closed 2019-05-04 by dustContributor

JulioJerez, 2019-05-04:
Ok, I will complete the player controller, in case you decide to take a
look and also because is time to re enable it again for any other user.
Player capsules is kind of a black art programming, is harder than it looks
to make good ones, I have written several for many games, so do not
underestimate the problem, is more than just moving a capsule.

JulioJerez, 2019-05-05:
ok this is the final comment, I added the basic player gameplay manipulation, now is can walk on the map control for the keyboard.
the functionality is very basic, by the controller is quiet sophisticated. 
from here is upto the end clinet application to add the game play functionality  the wish.

I have not tested it much, so if you try it out and find something you wish to be diffrent jut tell me, as these thing can alway be improve and the is not perfect solution. .

JulioJerez, 2019-05-05:
oh one more update,
I added a rolling terrain with a simple slope clip game play logic so that
you can see the kind of functionality I am talking about and how easy if to
set it up..