MADEAPPS/newton-dynamics #166: Crash in debug mode

TrevorCash, 2019-05-20:
Im getting this crash when my program starts in debug mode.

[image]

closed 2019-05-20 by TrevorCash

JulioJerez, 2019-05-20:
so what was the problem?

TrevorCash, 2019-05-20:
I was giving newton a 0.0 initial timestep

JulioJerez, 2019-05-20:
The estrange thing is that In Newton it used to be legal to take a time
step if zero length. This was how continue collision use to work bu is too
complicated.
This is disable for now, but when this functionality is on, a time step of
zero length is an impulsive step that only change the bodies velocities and
conserve momentum.
this has been disabled for some time now in favor or the update with sup
steps assuming the parallel solve can make up for it.
I am hoping that the GPU solver can realized that aspiration, by if is does
not the I will re enable  it.
But anyway glad you found the problem