MADEAPPS/newton-dynamics #169: Getting compound assert in debug mode with latest build

TrevorCash, 2019-05-20:
[image]

JulioJerez, 2019-05-21:
ah yes that's one optimization that I made over the weekend wit Dave.  It
is support to generate much better contact set for all collision shapes,
but I could no have the branch of the code trigger with compounds that I
have.
could you make if a favor and comment out the assert and let it go, and
tell me if the respond is better of worse than enabling the define
//#define DE_USE_OLD_CONTACT_FILTER
in file dgWorld.h?

JulioJerez, 2019-05-21:
Hi,
Also when the assert happens, if possible can you take a screen shot of the
scene I a curious as to what kind of collision configuration is producing
so many contacts. Basically I am trying to improve the quality of the
simulation by filtering the contact that aren't essential to form the
system matrix because the are linear combination of other contacts.
With simple convex this is easy, since the collision is a convex 2d
polygon, so any polygon simplification is better that nothing. For
compounds this is more difficult.

TrevorCash, 2019-05-21:
Hi Julio,  yeah I'll check it out when I get to my computer.

TrevorCash, 2019-05-21:
I got the assert again - but its hard to reproduce.  I can tell you though that the compounds look like this:
[image]

There are 3 boxes that form a compound shown by the blue lines if you look closely.  All other objects are simple shapes.

TrevorCash, 2019-05-21:
I can comment out the assert - so then it doesn't crash - but its hard to tell a behavioral difference with my setup.

JulioJerez, 2019-05-21:
umm that very unsettling, 
if you look at the lien whe the assert come from

							if (contactCount > (DG_MAX_CONTATCS - 2 * (DG_CONSTRAINT_MAX_ROWS / 3))) {
#ifdef DE_USE_OLD_CONTACT_FILTER
								contactCount = m_world->OldReduceContacts(contactCount, contacts, DG_CONSTRAINT_MAX_ROWS / 3, proxy.m_contactJoint->GetPruningTolerance());
#else
								dgAssert(0);
								contactCount = m_world->PruneContacts(contactCount, contacts, proxy.m_contactJoint->GetPruningTolerance(), 16);
#endif
							}
,

the temp buffer is 128 contacts minus the max contacts per colliding pair which is 16t, 
therefore the contact count has to be larger than 90 contacts for the function to be called. 
I do not think that shape can generate that many contact points unless the mesh under the 
L shape box is a very high resolution mesh, is that the case?  

but anyway before I remove the assert I will add a trace, just to make sure the call is trigger because of a large number of contact and no because of a bug. 
I almost have never seem that function to be trigger 

The contact reduction is call when the number  contacts calculate exceeded the maximum number on  the temp buffer capacity. 

In any case what the code does in the contacts final contacts set should be the extreme points of the convex hull of the set, about four or maybe five contacts which should produce a much higher quality collision resolution than if we for example 4 contacts for each leg since the interior contacts  
can will generate a force value that can be recreated by the vale force at the corners.

TrevorCash, 2019-05-21:
There are lots of bodies in the scene - about 70 bodies - is the temp buffer global?

JulioJerez, 2019-05-21:
no the temp buffer is per colliding pairs. But if a compound has many
shapes and the ground has many faces the it can generate many contacts.

TrevorCash, 2019-05-21:
ok - in my scene there are no complex bodies.  just basic shapes - cylinders and boxes.

JulioJerez, 2019-05-21:
that was my impression, what about the floor is the floor a high resolution mesh?
I would be interesting to see what causing all those contacts.

TrevorCash, 2019-05-21:
its actually a 2d array of boxes.  making up a compound - Ah that must be it - didnt think of that.

JulioJerez, 2019-05-22:
ah I see, assuming teh worse case say a teh L shape lie sideway, thsi mean all shape hit the floor, also assuming thet the contact count is 92 contacts, if each colliding pair or convex between the compounds  generate on average 6 contacts, this mean that 92 /  6 = 12 the l shape his approximately 12 box of teh compound floor seems on reasonable. 
I am goint to remove the assert 0, and commite the code.

Is there any reason you use a compound or boxes for floor, I liek the idea because it stress test the engine, in fact ther is a shape called scen collision  that is design for that. 

but for you case I leoive it will be far more effecine to use a poligonal mesh for floor.
I do not kwno if you know but the engine has a sofoticated polgon optimization proccess that simply teh collsion mesh to miximaxe the suface area of each face while keep the error wit the teh numerical precision of double, (praticall zero error between the origional and the collision mesh)
the collision mesh is very efficent for these cases.  
unless you are using the compound for some other reasons. 

.

TrevorCash, 2019-05-22:
yeah they are boxes only because thats how the code evolved - Really it should be a single flat plane or something.   As it is the boxes are part of the scene collision.

TrevorCash, 2019-05-22:
Oh Sorry I told you wrong.  I do have 100x100 boxes -  but there is only one box for the collision.  I looked at the code too quickly.  :/

JulioJerez, 2019-05-22:
100x100? does this mean one box of 100 unit by 100 unit or a compound of
100 x 100 small boxes? 🤔 hikes!!!

TrevorCash, 2019-05-23:
As far as the physics is concerned there is only 1 box that represents the ground.

JulioJerez, 2019-05-23:
Yes I know that, what I want to know is how many sub shapes are touching the aabb of the L shape body, that will explain the source of so many contact for a shape that seem very simple.
If there aren't enough sub shapes there is bug. And bugs have a way to show up.

TrevorCash, 2019-05-24:
got the assert in debug mode again - this time I was working with simple box shapes:
[image]

JulioJerez, 2019-05-24:
I removed all the assert from the file please sync and try again.

JulioJerez, 2019-05-24:
oh wait, you said you were playing with simplee boxes?
can I get a repro test, I am curious to find out why that is. I very rarerly seem those asserts.

TrevorCash, 2019-05-28:
here is a demo: https://drive.google.com/open?id=19czDAmXuzggpCeHHyOvdskDerV6CxmQO

JulioJerez, 2019-05-28:
It turned out there is a bug that does affect the posporcesing of of contacts generated by compounds, or more specifically contracts that do not lie on a plane.
I will fix that tonight and them test it with you demo.
It affect a user that just reported in the forum, so Icommitted a temp fix until I write a better volume filter.

TrevorCash, 2019-05-28:
Sounds Good, better compound behavior sounds great.  

Also let me know if you make a gitlab account, I'll add you to the my repository in case you want to build there.

JulioJerez, 2020-03-08:
I believe this was fioxed.

closed 2020-03-08 by JulioJerez