MADEAPPS/newton-dynamics #172: Assert when using small cylinders

TrevorCash, 2019-07-16:
Hi Julio,

I am getting an assert that only came up when I started adding small cylinders to my scene:
[image]

The pivot variable is very small (9.7e-7f) causing the assert.

I am using newton commit: 2fe6b0350adc8abda07198b2da48dd917efdd677

The commit on the project is f0ee1b9079af96cbe40e468a17b088899960ac55 (latest master)

Im having a hard time seeing what would cause the pivot to be small.

JulioJerez, 2019-07-16:
Oh the assert is in the wrong place.  this come from function void
dgDynamicBody::IntegrateImplicit(dgFloat32 timestep)

the is a error reduction pivoting technique, basically what is does is that
when inverting a or calculation a pivot (linear combination)
the numerical error if proportion the magnitude of the pivot.
the idea is that before pivoting, you want to have in the main signal the
column number with the largest absolute values.
in general is not a problem if the diagonal is reasonable large. by when
the values area all small you can get into the problem that   you get a
numerical solution that is no the solution of the system.

if you look at the variables, you see that i and permute have the same
value, so not need to permute, because even when the diagonal is small it
is still larger that all the element on the column.
I fixed by moving teh assert to inside the if (permute != i)
if the assert happen there, then some really wrong is happening, because it
mean the matrix is near singular. which can be fro a mass matrix.

please try sync.

TrevorCash, 2019-07-16:
I see, ok I'll pull from the latest.

closed 2019-07-16 by TrevorCash