MADEAPPS/newton-dynamics #174: Assert In dgBroadPhase::AttachNewContact

TrevorCash, 2019-07-16:
This assert only seems to come up when I have a cylinder object in my scene:

[image]

JulioJerez, 2019-07-16:
that looks wrong, somehow a joint is twice in the list, that a big problem.
there is nothing especial about cylinders.
it is reproducible?

TrevorCash, 2019-07-16:
I'll look into it further - weirdly - it is only happening when I attach a kinematics joint to the body.  again it only happens when the body is a cylinder and does not happen when the body is a compound in my test case.

I have not changed my contact parsing code in a long time - but Its possible there is a bug there.

I just pushed my project up too.  if I switch shapes to "gear_medium" shown in this code:
[image]

The assert happens consistently.  In the game you have to pick up the object with left click though.

Like you say though I don't think its the shape.  Because the NewtonPhysicsTests Demos work when you right click spawn the assortment of objects.

TrevorCash, 2019-07-18:
Hi Julio, 

I have been testing against various versions of newton the crash is always there - There are too many contacts.  

Sometimes I get the crash in dgWorld::FlushCache(), the function tried to delete all contacts but the contactList.m_contactCount is still > 0 which causes an assert.

I have tried commenting out my NewtonProcessCantacts callback, the issue is still there.  It only happens after I delete a body and create a new one.  Could it be I need to manually destroy contacts in my code?  so far I am letting newton handle it.

Thanks,
Trevor

TrevorCash, 2019-07-18:
After looking further it this behavioral only happens after the changes introduced between 

newton commit: 6f7d3bf1b7ffc030fb3a6e2fa588c561b13ea338 

and 

newton commit: ba44ab69b9667d68c33a0f0fe504062519542f43

Which makes sense because of the changes in dgContact.

JulioJerez, 2019-07-18:
I have not have time to check the error, but that last observation you made is a good clue to the reason of the bug.
What happen is that I made a changed that replaced the contact list with a contact array, this is faster but it lose the ability to delete contact in constant time.
So I added a flag the mark the contact dead.
But there may be a bad bug with deleting a body yes, I will check this Saturday.

TrevorCash, 2019-07-18:
Thanks a bunch!

JulioJerez, 2019-07-19:
in preparation to check these bug I clone your project again and using cmake x64 
with the default setting I only got one compile error left

1>------ Build started: Project: TechGame, Configuration: Debug x64 ------
1>ManipulationTool.cpp
1>c:\users\jjerez.bsg\documents\github\techgame\techgame\ManipulationTool.h(2): fatal error C1083: Cannot open include file: 'Urho3D/Urho3DAll.h': No such file or directory
1>NodeTools.cpp
1>c:\users\jjerez.bsg\documents\github\techgame\techgame\NodeTools.h(2): fatal error C1083: Cannot open include file: 'Urho3D/Urho3DAll.h': No such file or directory
1>Generating Code...
1>Done building project "TechGame.vcxproj" -- FAILED.
....
....
2>------ Skipped Build: Project: INSTALL, Configuration: Debug x64 ------
2>Project not selected to build for this solution configuration 
========== Build: 0 succeeded, 1 failed, 61 up-to-date, 1 skipped ==========

do you know who to fix that?
also does the code that is committed generates these bugs?

TrevorCash, 2019-07-19:
Yeah just add this switch in cmake:
-DURHO3D_MONOLITHIC_HEADER=ON

TrevorCash, 2019-07-19:
That enables the Urho3D/Urho3DAll.h header file.  (its an option because it is slower to compile with the combined header.  I use it for prototyping games though)

TrevorCash, 2019-07-19:
the full list of options I use is this:

```
-DURHO3D_CSHARP=OFF \
-DURHO3D_MONOLITHIC_HEADER=ON \
-DURHO3D_SAMPLES=ON \
-DURHO3D_WIN32_CONSOLE=ON \
-DURHO3D_WEBP=OFF \
-DURHO3D_NAVIGATION=OFF \
-DURHO3D_URHO2D=OFF \
-DURHO3D_NETWORK=OFF \
-DURHO3D_PROFILING=OFF \
-DNEWTON_WITH_AVX2_PLUGIN=ON \
-DNEWTON_DOUBLE_PRECISION=ON
```

JulioJerez, 2019-07-23:
I have a litle time to try this again. I set the options like you said and
I now get these errors that I never seen before.

9>Error HRESULT E_FAIL has been returned from a call to a COM component.
9>Error: Error HRESULT E_FAIL has been returned from a call to a COM
component.
10>------ Build started: Project: Assimp, Configuration: Debug x64 ------
10>Error HRESULT E_FAIL has been returned from a call to a COM component.
10>Error: Error HRESULT E_FAIL has been returned from a call to a COM
component.
11>------ Build started: Project: crnlib, Configuration: Debug x64 ------
11>Error HRESULT E_FAIL has been returned from a call to a COM component.
11>Error: Error HRESULT E_FAIL has been returned from a call to a COM
component.
12>------ Build started: Project: crunch, Configuration: Debug x64 ------
12>Error HRESULT E_FAIL has been returned from a call to a COM component.
12>Error: Error HRESULT E_FAIL has been returned from a call to a COM
component.
13>------ Skipped Build: Project: INSTALL, Configuration: Debug x64 ------
13>Project not selected to build for this solution configuration
========== Build: 0 succeeded, 12 failed, 74 up-to-date, 1 skipped
==========

I type some lines in google and I get various opinion but not a definite
answer as to what is it.

JulioJerez, 2019-07-23:
I am using vs 2017 professional.

TrevorCash, 2019-07-23:
hmm.  Can you try building just rbfx?  You can run cmake inside the rbfx folder and make a build directory inside that.  just to make sure the base game engine builds.  you can use default cmake parameters.  the Urho3d project should build

JulioJerez, 2019-07-23:
for completion I try generating a visual studio 2015 solution but does not
compile at all because
I believe it uses cpp features not available in  vs 2015.
or maybe something else. by the errors are too many

17>C:\Users\jjerez.BSG\Documents\GitHub\TechGame\rbfx\Source\Tools\Editor\Pipeline\Pipeline.cpp(160):
error C3538: in a declarator-list 'auto' must always deduce to the same type

JulioJerez, 2019-07-23:
when I do that is does compile yes. some warnings but I see the in many
other project too.
I guess the rbtx is the core urtho project,. because I do not see any of
the newton or the TexcGame projects.

TrevorCash, 2019-07-23:
Yes rbfx is a newer version of the Urho3D project.  the TechGame project has rbfx as one of the submodules, as well as other plugins like rbfx-newton, etc..

rbfx uses c++17.  

It is interesting you can build the rbfx project but not the entire TechGame project.  

It could be a visual studio versioning problem: I am using vs 2017 15.9.7.

also I am using cmake 3.12 but I dont think thats the problem.

TrevorCash, 2019-07-23:
if you want you can also test just the rbfx-newton plugin by making a folder structure like this:

MyTestSolution
->rbfx/...
->rbfx-newton/...
->CMakeLists.txt

then the resulting solution will have rbfx and rbfx-newton.

JulioJerez, 2019-07-23:
I suspect too this is a versioning shit from some sequrity nonsense.
I tryied o do this in me lunch time from work, But I can try tonight at home with the free version of vs 2017.

TrevorCash, 2019-07-23:
Yeah that might be it.  I am using community vs 2017.

JulioJerez, 2019-07-26:
I do npt not get why thsi has been so hard to get, I try to clone again form home by I get a GitHub error, that a sync point is missing, I can get the exact error because giuthub does no let you copy form the screen, by if I ignore the error and try to create the solution 
I get this error from  cmake.

CMake Error at CMakeLists.txt:14 (add_subdirectory):
  The source directory

    D:/TechGame/rbfx-openvr

  does not contain a CMakeLists.txt file.

TrevorCash, 2019-07-26:
Hi Julio,

It looks like somehow that submodule is not fully initialized.  if you checkout the Stable branch I just pushed to, all submodules should be in order.

JulioJerez, 2019-07-26:
this is what I see in teh github app..
> C:\Users\jjerez.BSG\Documents\GitHub\TechGame\rbfx
> C:\Users\jjerez.BSG\Documents\GitHub\TechGame\rbfx-newton
> C:\Users\jjerez.BSG\Documents\GitHub\TechGame\rbfx-openvr

> @ -1 +1 @@
> Subproject commit 9df35ac8d47cd1364831610eced4c38d96b8d86c
> Subproject commit da37977a6d21ac7af7a5a06600f03f24e271fe11
but is does not move pass that point even if I force to re clone. 

> if you checkout the Stable branch I just pushed to, all submodules should be in order.?
how do I do that?

TrevorCash, 2019-07-26:
git checkout Stable
git submodule init

of the top of my head.

TrevorCash, 2019-07-26:
Have you used a tool like GitKraken before?  it makes working with submodules a little easier.  all can be done with command line of course though.

TrevorCash, 2019-07-28:
From the root TechGame directory you can use: 
`git submodule update --init --recursive` if some submodules are not initialized
https://stackoverflow.com/questions/10168449/git-update-submodules-recursively

JulioJerez, 2020-03-08:
I believe this was fixed.

closed 2020-03-08 by JulioJerez