MADEAPPS/newton-dynamics #178: Adjacent NewtonCreateTreeCollision shapes

Bytenjoy, 2019-09-07:
I have a tiled world composed of many tiles that can be load and unloaded in realtime. Each tile has a NewtonCreateTreeCollision shape. There are a lot of adjacent tiles with contiguous and coplanar polygons. The problem is that when a ball rolls over these tiles, it often bumps at the union between two adjacent tiles (because of colliding with an edge, I guess). Is there something I can do at the application level to avoid the bumps? I'm using the Newton 3.14 release, not the latest commit.

Thanks.

Note: I wanted to ask this at the forums, but after two weeks from registering, my user hasn't been activated yet and this is the only way I found to ask.

charlesyulife, 2019-09-07:
I have also peoblem with the treecollider,i'm using unity,and i want a dynamic treecollider which Newton not surpport, maybe you can help me with that?

Bytenjoy, 2019-09-07:
I only know that NewtonCreateTreeCollision makes the body behave as a static body. In my case, the tiles are static, so I don't have problem with that.

JulioJerez, 2019-09-07:
I had that problem too when doing the player controller, and it was fixed, 
Try getting the heard revision, there has been lot of work on the contacts generation to address that problem.

JulioJerez, 2019-09-07:
on the forum registration I have to suspend registration, some how there is a person on the internet that decided to attack my forum by registering over 100 account every day to post random links.
I am not sure what the win by that, but if the goal was to shoot me down they had succeeded.

charlesyulife, 2019-09-08:
@JulioJerez That's possibly out of envy and commercial purpose! Because NewtonPhysics is different and outstanding, it can do what the giant engine can't!   If this let you down,what a pity!  For one person but prevented thousands of people to join，that's unworthy! If this guy bother you,i can help you to clear garbage infomation from the person. i'm realy mean it and i hope i can join the team to work with you begin from clearing the bad guy.

NewtonPyisics has  amolst save my day, only if it support dynamic treecollider with continuous collision detection,better in unity!I have post mu repo on github and you can see that,i'm using it do the real physic simulation! could you please help me and let me help you too!

charlesyulife, 2019-09-08:
@JulioJerez Fyi Also we have an academic and business community with a lot of high achievers.I will help you promote newtonPhysics, so that more real users will know about it and help its development. @Bytenjoy Sorry to borrow your site bro.

Arbos, 2019-09-09:
I tried the latest commit and the bumps still happen. I'll explain a particular test I'm doing, in case I didn't explain well. I have a row of cubes, where each cube is a static body with a tree collision shape. When a ball rolls over the cubes, it often bumps at the edges of the cubes. Is this supposed to be fixed? If it's not fixed, can I do something at the application level? Like, for example, filter the edge contacts with NewtonMaterialSetCollisionCallback.

Bytenjoy, 2019-09-09:
I tried the latest commit and the bumps still happen. I'll explain a particular test I'm doing, in case I didn't explain well. I have a row of cubes, where each cube is a static body with a tree collision shape. When a ball rolls over the cubes, it often bumps at the edges of the cubes. Is this supposed to be fixed? If it's not fixed, can I do something at the application level? Like, for example, filter the edge contacts with NewtonMaterialSetCollisionCallback.

Bytenjoy, 2019-09-09:
Here a video showing the bumps:

https://youtu.be/p4nMVzlL8b4

JulioJerez, 2019-09-10:
if each segment a separate collision tree?
that's not going to work, edge information is only calculate for edge of
the same collision mesh.
and the edge have to be a single manifold. meaning each edge can only be
shared by one of two faces.

you can try setting the collision margin of the ball, so that the surface
does no touche the  surface. but I have never tried that.

JulioJerez, 2019-09-10:
another trick you could do is to make the boxed overlap a lithe,
this way when teh ball is going to go over and edge, wit will still have
some flat face support
this is more of less how edge information workd.

Bytenjoy, 2019-09-10:
Yes, each segment is a separate collision tree because each segment can be added or removed from the world in realtime. If all the segments were a single collision tree, each time I add or remove a segment, the collision tree BVH, etc. would have to be recalculated (If I'm not wrong), which is not suitable for realtime for a large number of polygons.

I'll try your suggestions and if they don't work then I'll have to redesign this part of the game.

closed 2019-09-10 by Bytenjoy