MADEAPPS/newton-dynamics #184: Compilation Issue With VS 2019

TrevorCash, 2019-09-18:
Hi Julio,

I upgraded to vs 2019 and tried compiling the demos-sandbox.  I am getting this error:

[image]

JulioJerez, 2019-09-18:
that's a serious bug, no way the tread ID does no match the id of the caller thread onlease teh parameter threadData is wrong.

```
void* dgThread::dgThreadSystemCallback(void* threadData)
{
	dgFloatExceptions exception;
	dgSetPrecisionDouble precision;

	dgThread* const me = (dgThread*) threadData;
	D_SET_TRACK_NAME(me->m_name);
	me->Execute(me->m_id);
	dgInterlockedExchange(&me->m_threadRunning, 0);
	return 0;
}
```

I hope MS did add some shenanigan  to the thread callback user data object.
I do not have VS 2019, I will try tonight.
is this the free version?

TrevorCash, 2019-09-18:
Yeah I thought it was an odd bug to see a threading bug like this pop up.  yeah its the free (community) version

JulioJerez, 2019-09-18:
of topic, can you build on lower version of Visual studio? is so has you
seem the work on the robot manipulators.
I though you might be interested on that sort of thing? it is is all
Physics by the book.

TrevorCash, 2019-09-18:
yeah the reason I tried vs 2019 is because rbfx is starting to move to 2019 due to c++14 features.  yeah I want to check out the demos.  do the manipulators works with the standard joints?  yeah Robotics in tech game is what I want to move towards.

JulioJerez, 2019-09-18:
all plain and simple hinges yes, no more esoteric inverse IK  joints.
it does uses a listener which is encapsulate in a new library called dModel.
this new dModel library will superseded all the competitive that is in the
dCustomJoint and do no belong there.

the definition of a dModel in newton is a collection of of hiding bodies,
collision and joint wrapped under a listener.
they have a pre and post update the the client app can use to  apply
control to the model.

this will be the base call for vehicles and animated characters.
next I will refactor the hexapod and teh I will try to make eighth the TRed
of a AT-ST like robot, cross my finger knock on wood.

TrevorCash, 2019-09-18:
A single manager makes sense.  Ill have to try that out.

TrevorCash, 2019-09-18:
Just tried the SixAxisManipulator Demo - Super Cool

DaveGravel, 2019-09-19:
Hi TrevorCash, I know this is not the solution for the problem, But if you absolutly need to build newton for work on your project with 2019 you can uncomment //#define DG_USE_THREAD_EMULATION in dgTypes.h file.
On this way you can continue to work with your project from 2019 in waiting the real fix.

TrevorCash, 2019-09-19:
Hi Dave, Great!  Thanks for letting me know!

DaveGravel, 2019-09-20:
Hi TrevorCash.
If you try again with last sdk, 
You don't need to uncomment //#define DG_USE_THREAD_EMULATION anymore.
The problem is fixed with 2019 now. 
I have do some tests and it work good for me.

closed 2019-09-20 by TrevorCash